olikraus/u8g2 #432: u8g2 in 128x64 ST7920 using STM32F0

LuizRoque, 2017-11-24:
Hi, I've been trying to use the u8g2 in STM32F070 - nucleo board and nothing is writing on display. I tried using arduino and worked perfect!

I connect in the following Pins,

PB11 - CLK SPI - E
PB13 - MOSI SPI - R/W
PB1 - RS
PB2 - RST

The PSB I pull to ground.

Here is my whole code.

`/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "u8g2.h"

// LCD 128x64
u8g2_t u8g2;

uint8_t count = 0;

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void Error_Handler(void);
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr);

/* Private functions ---------------------------------------------------------*/

/**
  * @brief  Main program.
  * @param  None
  * @retval None
  */
int main(void)
{

  HAL_Init();

  /* Configure LED2 */
  BSP_LED_Init(LED2);
  /* Configure the system clock to 48 MHz */
  SystemClock_Config();

  u8g2_Setup_st7920_128x64_f(&u8g2, U8G2_R0,
		  u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32);  // init u8g2 structure

  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0); // wake up display

  HAL_Delay(100);

  u8g2_ClearDisplay(&u8g2);
  u8g2_SetFont(&u8g2, u8g2_font_5x7_tf);

  u8g2_ClearBuffer(&u8g2);
  u8g2_DrawStr(&u8g2, 0, 24, "Hello world!");
  u8g2_SendBuffer(&u8g2);

  /* Infinite loop */
  while (1)
  {

  }
}

uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	GPIO_InitTypeDef  GPIO_InitStruct;

	switch(msg){
		//Initialize SPI peripheral
		case U8X8_MSG_GPIO_AND_DELAY_INIT:

			__HAL_RCC_GPIOB_CLK_ENABLE();
			/* SPI SCK, MOSI GPIO pin configuration  */
			GPIO_InitStruct.Pin = GPIO_PIN_13 | GPIO_PIN_15;
			GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
			GPIO_InitStruct.Pull = GPIO_NOPULL;
			GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
			HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

			//RST_LCD_GPIO_CLK_ENABLE();
			GPIO_InitStruct.Pin = GPIO_PIN_2;
			GPIO_InitStruct.Pull = GPIO_NOPULL;
			GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
			HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

			//CS_LCD_GPIO_CLK_ENABLE();
			GPIO_InitStruct.Pin = GPIO_PIN_1;
			GPIO_InitStruct.Pull = GPIO_NOPULL;
			GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
			HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

		break;

		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);

		break;
		//Function which delays 10us
		case U8X8_MSG_DELAY_10MICRO:
		for (uint16_t n = 0; n < 320; n++)
		{
			__NOP();
		}

		break;
		//Function which delays 100ns
		case U8X8_MSG_DELAY_100NANO:
		__NOP();

		break;
		//Function to define the logic level of the clockline
		case U8X8_MSG_GPIO_SPI_CLOCK:
			if (arg_int) HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOB, GPIO_PIN_13, GPIO_PIN_RESET);

		break;
		//Function to define the logic level of the data line to the display
		case U8X8_MSG_GPIO_SPI_DATA:
			if (arg_int) HAL_GPIO_WritePin(GPIOB, GPIO_PIN_15, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOB, GPIO_PIN_15, GPIO_PIN_RESET);
			BSP_LED_Toggle(LED2);

		break;
		// Function to define the logic level of the CS line
		case U8X8_MSG_GPIO_CS:
			if (arg_int) HAL_GPIO_WritePin(GPIOB, GPIO_PIN_1, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOB, GPIO_PIN_1, GPIO_PIN_RESET);

		break;
		//Function to define the logic level of the Data/ Command line
		case U8X8_MSG_GPIO_DC:

		break;
		//Function to define the logic level of the RESET line
		case U8X8_MSG_GPIO_RESET:
			if (arg_int) HAL_GPIO_WritePin(GPIOB, GPIO_PIN_2, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOB, GPIO_PIN_2, GPIO_PIN_RESET);
			BSP_LED_Toggle(LED2);

		break;
		default:
			return 0; //A message was received which is not implemented, return 0 to indicate an error
	}

	return 1; // command processed successfully.
}

/**
  * @brief  This function is executed in case of error occurrence.
  * @param  None
  * @retval None
  */
static void Error_Handler(void)
{
  while(1)
  {
    /* Turn LED2 on */
    BSP_LED_Toggle(LED2);
    HAL_Delay(1000);
  }
}

/**
  * @brief  System Clock Configuration
  *         The system Clock is configured as follow : 
  *            System Clock source            = PLL (HSI)
  *            SYSCLK(Hz)                     = 48000000
  *            HCLK(Hz)                       = 48000000
  *            AHB Prescaler                  = 1
  *            APB1 Prescaler                 = 1
  *            HSI Frequency(Hz)              = 8000000
  *            PREDIV                         = 2
  *            PLLMUL                         = 12
  *            Flash Latency(WS)              = 1
  * @param  None
  * @retval None
  */
void SystemClock_Config(void)
{
  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_OscInitTypeDef RCC_OscInitStruct;
  
  /* No HSE Oscillator on Nucleo, Activate PLL with HSI as source */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_NONE;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PREDIV = RCC_PREDIV_DIV2;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL12;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }

  /* Select PLL as system clock source and configure the HCLK, PCLK1 clocks dividers */
  RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1);
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }
}`

Any help would be appreciated.

Luiz

olikraus, 2017-11-24:
At least I can not help here. It also does not make sense to post the same code twice.

LuizRoque, 2017-11-25:
Hi, I correct the code using the guide in
https://github.com/olikraus/u8g2/issues/356
And worked! But there is a bar in the display.
[image]

You know what could be? How I clean that bar.

olikraus, 2017-11-28:
looks like the picture is shifted down. Either the communication is not stable or the U8g2 driver does not fit to your display.

olikraus, 2017-12-07:
not sure, how to support here...

closed 2017-12-07 by olikraus

Andromeduch, 2018-02-07:
Top bar solution in [#487](https://github.com/olikraus/u8g2/issues/487)