olikraus/u8g2 #434: Two/multiple OLEDs with u8g2 - small problem

walenw, 2017-11-26:
Dear Friends,
Sorry for bothering you with this simple issue, but it is connected with this thread. I have two OLED SH1106 displays connected via I2C. I have changed physical address resoldering apropiate pins on the bottom side of displays. Now I have two devices on I2C bus - 0X3C and 0X3D. The problem is that I can not use two displays the same time. It is not working. What can I do?

[image]

`#include "U8g2lib.h"
#include <Wire.h> //Library for I2C interface 

U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_1(U8G2_R0, U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_2(U8G2_R0, U8X8_PIN_NONE);

void setup() {
  u8g2_SetI2CAddress(&OLED_1, 0x3C);
  u8g2_SetI2CAddress(&OLED_2, 0x3D);
  OLED_1.begin();
  OLED_2.begin();
  OLED_1.setFont(u8g_font_6x10);
  OLED_2.setFont(u8g_font_6x10);
}

void loop() {
  static unsigned long oledTimer = millis();      //every 1000ms update the oled display
  if (millis() - oledTimer >= 1000)  {
    oledTimer = millis();
    drawOLED_1();
    //drawOLED_2();
  }
}

void drawOLED_1(void) {
  char buffer[10];
  OLED_1.clearBuffer();          // clear the internal memory
  OLED_1.setFont(u8g_font_6x10);
  OLED_1.drawStr(0, 10, "Sensor 1 -> ");
  dtostrf(10, 5, 0, buffer);   //print value 10
  OLED_1.drawStr(80, 10, buffer);
  OLED_1.sendBuffer();         // transfer internal memory to the display
}

void drawOLED_2(void) {
  char buffer[10];
  OLED_2.clearBuffer();          // clear the internal memory
  OLED_2.setFont(u8g_font_6x10);
  OLED_2.drawStr(0, 10, "Sensor 2 -> ");
  dtostrf(20, 5, 0, buffer); //print value 20
  OLED_2.drawStr(80, 10, buffer);
  OLED_2.sendBuffer();         // transfer internal memory to the display
}`

olikraus, 2017-11-28:
Try

u8g2_SetI2CAddress(&OLED_1, 0x3C*2);
u8g2_SetI2CAddress(&OLED_2, 0x3D*2);

See: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

Oliver

walenw, 2017-11-28:
Nothing has changed. Moreover when I apply:
 u8g2_SetI2CAddress(&OLED_1, 0x00);
 u8g2_SetI2CAddress(&OLED_2, 0x00);
result is not changing.

After uncommenting the OLED_2
void loop() {
static unsigned long oledTimer = millis(); //every 1000ms update the oled display
if (millis() - oledTimer >= 1000) {
oledTimer = millis();
drawOLED_1();
drawOLED_2();
}
The screen is blinking every 1 second.

olikraus, 2017-11-29:
There was a formating problem in my previous post. You must multiply the address with 2:

```
u8g2_SetI2CAddress(&OLED_1, 0x3C*2);
u8g2_SetI2CAddress(&OLED_2, 0x3D*2);
```

See: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

walenw, 2017-11-29:
I did it already.  The program is not responding on i2C address change. That is the problem. 
In general nobody is thinking about changing the IiC address in common OLED application. People just take a display and use it without thinking about i2C. In my case it is important. May be there is a bug in my u8g2 lib

olikraus, 2017-11-29:
Ah, another issue, you should use

```
u8g2.setI2CAddress( 0x3C*2);
u8g2.setI2CAddress(0x3D*2);
```

walenw, 2017-11-29:
Problem SOLVED!

Thank you olikraus:)

Below you can see working code.

`#include "U8g2lib.h"
#include <Wire.h> //Library for I2C interface

U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_1(U8G2_R0, U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_2(U8G2_R0, U8X8_PIN_NONE);

void setup() {
  OLED_1.setI2CAddress(0x3C * 2);        //<- SOLVED it works
  OLED_2.setI2CAddress(0x3D * 2);        //<- SOLVED it works
  //u8g2_SetI2CAddress(&OLED_1, 0x3C*2); This did not work for me
  //u8g2_SetI2CAddress(&OLED_2, 0x3D*2); This did not work for me
  OLED_1.begin();
  OLED_2.begin();
  OLED_1.setFont(u8g_font_6x10);
  OLED_2.setFont(u8g_font_6x10);
}

void loop() {
  static unsigned long oledTimer = millis(); //every 1000ms update the oled display
  if (millis() - oledTimer >= 1000) {
    oledTimer = millis();
    drawOLED_1();
    drawOLED_2();
  }
}

void drawOLED_1(void) {
  char buffer[10];
  OLED_1.clearBuffer(); // clear the internal memory
  OLED_1.setFont(u8g_font_6x10);
  OLED_1.drawStr(0, 10, "Sensor 1 -> ");
  dtostrf(10, 5, 0, buffer); //print value 10
  OLED_1.drawStr(80, 10, buffer);
  OLED_1.sendBuffer(); // transfer internal memory to the display
}

void drawOLED_2(void) {
  char buffer[10];
  OLED_2.clearBuffer(); // clear the internal memory
  OLED_2.setFont(u8g_font_6x10);
  OLED_2.drawStr(0, 10, "Sensor 2 -> ");
  dtostrf(20, 5, 0, buffer); //print value 20
  OLED_2.drawStr(80, 10, buffer);
  OLED_2.sendBuffer(); // transfer internal memory to the display
}`

olikraus, 2017-12-02:
Just to complete this:
```
//u8g2_SetI2CAddress(&OLED_1, 0x3C * 2); This did not work for me
```

This statement is wrong, because u8g2 expects something of type u8g2 struct, however you pass the address of a C++ class. This is illegal and breaks the code. It is required to use u8g2.GetU8g2() to get the address of the u8g2 structure first. The correct statement would look like this:
```
u8g2_SetI2CAddress(OLED_1.GetU8g2(), 0x3C * 2); This did not work for me
```

closed 2017-12-02 by olikraus