olikraus/u8g2 #435: ESP12E/F to SPI_HW SH1106

999chris, 2017-11-26:
Hi All,

I'm really concerned I'm doing something dumb here.

I have my SH1106 1.3" screen using 4W HW SPI connected to an ESP8266 module:

14 - CLK
13 - MOSI
15 - CS
12 - DC

My constructor:

999chris, 2017-11-26:
Okay I have this working under Software SPI, why is it not working with Hardware SPI? I've included the SPI.h library.

olikraus, 2017-11-28:
Maybe the ESP8266 HW SPI is available on other pins only. 
This is not an u8g2 issue, but probably more a question to your board, where the HW pins are located.

999chris, 2017-11-28:
Maybe, the CLK and MOSI pins match up, but I'm using the pin that is assigned for MISO for another purpose (CS). Could this be causing the issue?

olikraus, 2017-12-02:
could be...

olikraus, 2017-12-09:
not sure how to support here... closing...

closed 2017-12-09 by olikraus