olikraus/u8g2 #436: Right part of the screen have some random pixel

Eternyt, 2017-11-26:
[image]
No matter what I do, initDisplay() or begin(), this little bar remain whatever I do after.
Aliexpress ssd1306 128*64 using hardware I2C bus. Used the following setup:
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);

olikraus, 2017-11-28:
this is a sh1106 controller

Eternyt, 2017-11-28:
Thanks, fixed.
Just for curiosity, how do you spot that my screen is a sh1106 and not a ssd1306?

closed 2017-11-28 by Eternyt

olikraus, 2017-12-02:
> Just for curiosity, how do you spot that my screen is a sh1106 and not a ssd1306?

because of your title: " Right part of the screen have some random pixel"
:)