olikraus/u8g2 #437: Scroling mensage

pppeixoto, 2017-11-27:
Hello

My name is Pedro Peixoto and I’m trying to use U8G2 library to print a scrolling message on  ER-OLED018-1 S LCD with ssd1326 controller.
Code:
short x=0;
short offset=0;			// current offset for the scrolling text
short width;			// pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
const char *text = "teste";	// scroll this text from right to left

void setup(void) {

  /* U8g2 Project: SSD1306 Test Board */
  //pinMode(10, OUTPUT);
  //pinMode(9, OUTPUT);
  //digitalWrite(10, 0);
  //digitalWrite(9, 0);		

  /* U8g2 Project: T6963 Test Board */
  //pinMode(18, OUTPUT);
  //digitalWrite(18, 1);	

  /* U8g2 Project: KS0108 Test Board */
  //pinMode(16, OUTPUT);
  //digitalWrite(16, 0);	

  u8g2.begin();  
  
  u8g2.setFont(u8g2_font_inb30_mr);	// set the target font to calculate the pixel width
  width = u8g2.getUTF8Width(text);		// calculate the pixel width of the text
  u8g2.setAutoPageClear(1);
  u8g2.setFontMode(0);		// enable transparent mode, which is faster
}

void loop(void) {
  //delay(1000);  
  u8g2.firstPage();
   x=offset;
  do {    
     u8g2.drawGlyph(x, 30,'B');
     u8g2.drawGlyph(x+25, 30,'R');
     // u8g2.drawUTF8(x, 30, text);			// draw the scolling text    
  } while ( u8g2.nextPage() );
// scroll by one pixel
offset+=10;
//if (offset==150)
//offset = 0;  
 delay(100);							// do some small delay  
}
The problem is when the message is bigger than de display nothing is displayed, when the message is shorter is displayed but when reaches the end off display is stops working.
Any sugestions ?
Thanks

olikraus, 2017-11-28:
Did you activate u8g2 16 bit mode?

pppeixoto, 2017-11-28:
hello
i don't think so. How can i do it.
Thanks for the help

olikraus, 2017-11-29:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

closed 2017-12-07 by olikraus