olikraus/u8g2 #438: profont17 seems much wider than other Profont sizes

MattMatic, 2017-11-29:
Profont 15 and Profont 22 look lovely.
Profont 17 looks like 22's width but 17's height. i.e. it looks very widely spaced with too much gap.

Have tried both the _mf and _tf variants.

olikraus, 2017-11-29:
Yes, agreed, but I Just took the source files

MattMatic, 2017-11-30:
Ah. I've looked with FontForge at the BDF files. The metric font-width is wrong for Profont 17. Profont 15 has a width of 7, Profont 22 a width of 12. The width _was_ set to 14! Doh! Adjusted it to 9 for all glyphs (as monospaced anyway).

Shall rebuild and convert to C...

olikraus, 2017-12-02:
Can you attach the updated BDF or send a pull request for it?

MattMatic, 2017-12-04:
Sure :)
Can you rebuild the set of C files please? (I built a test one with the BDFCONV tool, but wasn't 100% sure of the parameters needed). I've included a screenshot of the ProFont series with the dud and the fixed 17.
Matt
[image]

[ProFont17-fix.bdf.zip](https://github.com/olikraus/u8g2/files/1526337/ProFont17-fix.bdf.zip)

olikraus, 2017-12-04:
thanks, i have replaced profont17.bdf

ToDo: New font file construction required

MattMatic, 2017-12-05:
Just rebuilt the fonts with the Mac. Here's the ProFont individual fonts (in case anyone's in need 😄 )
[u8g2_font_profont17.zip](https://github.com/olikraus/u8g2/files/1531114/u8g2_font_profont17.zip)

olikraus, 2017-12-09:
done...

closed 2017-12-09 by olikraus