olikraus/u8g2 #439: Render u8g2 full frame to file

MattMatic, 2017-11-30:
My embedded device is powerful, and I would like a method to save the full-frame buffer to either TGA or PBM (binary) format file, or to stream it over comms to render on the PC (as a mirrored display).

I'd rather not use any additional memory (RAM is tight)
Is it possible to do a GetPixel for a full-frame? I'm trying to get my head around the internal structure (FWIW, I'm using two drivers - the UC1610 and ST7565 if that helps).

PBM binary would be fine. i.e. for UC1610 : ASCII "P4\n160\n104\n" then binary row data, left to right, with MSbit being left-most (padded out to 8-bits multiple per row).

Thank you!

olikraus, 2017-12-02:
You could follow the idea here: 
https://github.com/olikraus/u8g2/tree/master/sys/bitmap
This will implement a new device, which handles the bitmap construction in parallel. Drawback is, that you need the frame buffer twice.

The other option should be to use the direct buffer API:
https://github.com/olikraus/u8g2/wiki/u8g2reference#direct-access-buffer-api
You can read the memory, rearrange the bits for PBM and write it to destination.

MattMatic, 2017-12-04:
Thanks Oli.
After a bit of puzzling, I sorted the getBufferPtr direct access API method. I misunderstood the comment above the "Memory Structure for controller with U8x8 support" - I first assumed that the memory structure would be different : "The internal pixel buffer has the same structure as the target display. Any modification of the internal pixel buffer will not be portable...". I thought the direct API would be in controller-byte-formation. Thankfully not :)

So, it was just a simple loop to create rows, top-to-bottom, left-to-right, rearranging the bits. No problem! 

Thanks,
Matt

MattMatic, 2017-12-04:
Pseudo code:

```
f = OpenFile(filename);
width = u8g2_GetBufferTileWidth() * 2;
height = u8g2_GetBufferTileHeight() * 2;
FilePrintf(f, "P4\n%d\n%d\n", width, height);
uint8_t *buff = u8g2_GetBufferPtr(&u8g2);
y = 0;
while (y < height)
{
 uint8_t ym = 1 << (y % 8); // Bitmask
 uint8_t *bp = &buff[(width * (y >> 3))]; // Start pointer of the row
 x = 0;
 while (x < width)
 {
  uint8_t d=0;
  if (bp[0] & ym) d |= 0x80;
  if (bp[1] & ym) d |= 0x40;
  if (bp[2] & ym) d |= 0x20;
  if (bp[3] & ym) d |= 0x10;
  if (bp[4] & ym) d |= 0x08;
  if (bp[5] & ym) d |= 0x04;
  if (bp[6] & ym) d |= 0x02;
  if (bp[7] & ym) d |= 0x01;
  WriteByteToFile(f, d);
  x += 8;
  bp += 8;
 }
 y++;
}
CloseFile
```

Obviously this can be optimised, but it at least shows you the right ordering.

NOTE: the resulting PBM bitmap is correct for U8G2_R0 orientation. When other rotations are used, the resulting bitmap is likewise rotated. e.g. U8G2_R2 will have an upside down bitmap.

olikraus, 2017-12-04:
> I thought the direct API would be in controller-byte-formation.

It IS in controller-byte orientation. But most controller seem to share the same memory orientation or can be configured into the same memory layout.

closed 2017-12-07 by olikraus