olikraus/u8g2 #440: ST7565R JLX12864-378G V2.0

Upcycle-Electronics, 2017-11-30:
Hello Oliver,
  I'm new to Git so bare with me please. I tried to test a JLX12864G-378 V2.0 display module using your library. The datasheet is mostly in Chinese, but it appears to use the ST7565R controller. I tried serveral different constructors for the ST7565 and ST7567. None of them are perfect. The list of constructors I used to test the display are here
`//The following were tested with the JLX12864G-378 V2.0 and pictures sent
//U8G2_ST7565_ZOLEN_128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7567_JLX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7565_ERC12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7565_LM6059_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7565_EA_DOGM128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7565_64128N_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
//U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 12, /* dc=*/ 10, /* reset=*/ 9);
`
[image]
This LCD is used on a lot of AVR Transistor Tester clones from China. I thought I'd pass along the info if you ever want to support it in your library. The datasheet for the display is found [here:](http://www.jlxlcd.cn/UpFile/201761584327.pdf)
-Jake

olikraus, 2017-12-02:
Then you should use the NHC_C12864 constructor and activate flip mode.

Upcycle-Electronics, 2017-12-02:
No problem.

olikraus, 2018-04-09:
Create an alias for the NHD_C12864: ST7565_JLX12864

olikraus, 2018-04-14:
Alias created...

closed 2018-04-14 by olikraus