olikraus/u8g2 #441: Fuzzy Intermittent Display

iotexpert, 2017-11-30:
Oliver,

I have several occasions where the screen will work... then Ill do a reset and they screen will have crazy random dots.  I am sure that it is something to do with the I2C interface... to fast or to slow... or not enough delay..

Has anyone seen anything like this?

Alan

olikraus, 2017-12-02:
Without more background information (code, pictures), this is difficult to say.

iotexpert, 2017-12-03:
I figured it out.  I have been using an I2C display with only vcc/gnd/scl/sda .. .I plugged in that also had the other SSD1306 pins pulled out.  One of which was reset which I just tied to VCC.  Well.. it turns out that you cannot de-assert the reset for some amount of time after powering up the chip.  It sucks because the chip responds to I2C commands... but the display is flaky.

closed 2017-12-03 by iotexpert