olikraus/u8g2 #442: Mirror/ flip a bitmap on its center axis

arjhun, 2017-12-02:
A common technique to create extra frames for sprites without using extra memory is by flipping a frame over it's axis. Because this wasn't implemented as far as i know in u8g2 i did it myself. I found it very useful for animating loops etc. I created a funciton called:

```
void setBitmapFlipMode(uint8_t flip_x, uint8_t flip_y)
      { u8g2_SetBitmapFlipMode(&u8g2, flip_x, flip_y); }
```

where x or y is true or false for mirroring/ flipping. And then the x and y start at x+=len and y+=len and get drawn x-- and y--. I don't know if this could be the way to do this, but I think it could be a great addition to the library!

olikraus, 2017-12-02:
well, yes... but a modification of the existing could should be avoided. Instead it is better to have a 
drawMirrorBitmap()
which gets some extra arguments for mirroring

Additionally: For adding this to u8g2, I need a working example and a description (similar to drawBitmap(), see the reference)

arjhun, 2017-12-02:
Yes that would be a nice way to go... I'll do a pull request when it's finished!

olikraus, 2017-12-02:
👍

closed 2017-12-30 by olikraus