olikraus/u8g2 #443: odd behavior with xor (color 2) over text

soundscu, 2017-12-02:
Hi,

I'm using the u8g2 port to pic33 that rbresca helped with a few weeks ago (https://github.com/olikraus/u8g2/issues/376).

First, I write a simple 5-line text menu to the screen, white letters on black background.
Then I draw a color 2 (xor) box around the menu option I want to highlight.

If my declaration of u8g2 (this line: u8g2_t u8g;) is in the main loop as a local variable, all works perfectly.

But if I move that declaration out of main to be a global variable, then the xor draws as blocks in the outline of the characters, rather than complete readable letters.  There are no compile errors.  Either way,  looking in the watch window, I can see that the u8g2 structure is in the exact same memory location and is the same size.

What I think must be a valuable hint:
If I compile with the global declaration, and program the pic, the code them runs immediately with blocks in inverse.
If I cycle the power for a clean restart, it remains displaying blocks.
[image]

But if I then compile for the local declaration, and program the pic, the code runs immediately AND STILL SHOWS BLOCKS.
After I cycle the power for a clean restart, I then have correct display of inverse characters.
[image]

Of course, I need u8g2 to be global so that I can have libraries of menu writing functions in separate files, not in the Main loop.

Attached is my project, you'll see some notes in newmainXC16.c.  Also attached are a couple of photos.
[U8G2_PIC33.X.zip](https://github.com/olikraus/u8g2/files/1524056/U8G2_PIC33.X.zip)

Many thanks for any help you can offer to solve this!
Thanks,
Jim

olikraus, 2017-12-03:
Did you set https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode to transparent?

soundscu, 2017-12-03:
That fixed it.  Many thanks!

In hopes of understanding this more deeply, why did declaring u8g2 locally seem to override the need to set fontmode to transparent?  Put another way, why is is necessary to setfontmode transparent only if I have declared u8g2 globally?

Thanks again.

olikraus, 2017-12-04:
hmm... yes indeed. it is not correctly set during init phase.

closed 2017-12-09 by olikraus