olikraus/u8g2 #444: Wrong results for u8g2.getStrWidth and u8g2.getUTF8Width for strings longer than 31 bytes

v1993, 2017-12-03:
Minimal code to reproduce this bug:

    char chars[] = "                                ";
    String str = chars;
    
    Serial.println(str.length()); // 32
    u8g2.setFont(u8g2_font_t0_15b_tf); // Or any other valid font
    Serial.println(u8g2.getUTF8Width(chars)); // 0
    Serial.println(u8g2.getStrWidth(chars)); // 0
In fact, it resets on every 32's byte. So `width for 1 char == width for 33 chars == width for 65 chars` etc.

olikraus, 2017-12-03:
This is not a bug. All coordinates, length and position information are 8 bit by default. This will lead to a "reset" at 256.
I guess width == 8 for 1 char. So for 32 chars you have 8*32 = 256, which is 0.

You can enable 16 bit mode (see FAQ) if you need 16 bit precision.

v1993, 2017-12-03:
Oh, thanks. It really works.

closed 2017-12-03 by v1993

olikraus, 2017-12-03:
:)