olikraus/u8g2 #445: ESP32 HW I2C not work

kiooer, 2017-12-04:
Hi,
I'm using the u8g2 port to SSD1309 128*64.
I try to use function 
`//U8G2_SSD1309_128X64_NONAME2_1_SW_I2C u8g2(U8G2_R0,/* clock=*/ 13, /* data=*/ 12, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0,/* clock=*/ 13, /* data=*/ 12, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1309_128X64_NONAME2_1_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE,13,12);
U8G2_SSD1309_128X64_NONAME2_F_2ND_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE);`

When i chose to use "SW" function ,it working well.
But "HW" function default.
How can i solve this problem?
There have same problem in issues #385 .
In issues #385 blotfi got work well ,but i'm unlucky.
Is there any details i missing?

olikraus, 2017-12-04:
what is the exact hw constructor you have used?

kiooer, 2017-12-04:
I use :
HW:
`U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE,13,12)`
SW:
`U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0,/* clock=*/ 13, /* data=*/ 12, /* reset=*/ U8X8_PIN_NONE)`
With Arduino IDE 1.8.5 and U8g2 latest version.

olikraus, 2017-12-04:
And you say, that SW I2C works, but HW I2C does not work, correct?

kiooer, 2017-12-04:
I need more speed on my project,maybe i should use SPI to connect OLED.
But i also not sure u8g2 whether this is supported by ESP32.
According to #187 ,i need use u8g2_esp32_hal.h to configure transmit Pin,but isn't support ESP32 Ardunio Core.

olikraus, 2017-12-04:
Actually I do not know how to support here. Can you mesure the pins? Do you see activity there? Maybe also speed is too high for the OLED. How long are your wires? Did you use proper pullup resistors?

kiooer, 2017-12-06:
I tried to use 4.7K&10K resistors to pullup SCL and SCA yesterday,but ESP32 loop the error message and can't initialize.
Today i will link logic analyzer to check SCL&SCA Pins.
My wires's length is less than 5cm.

JamesGKent, 2017-12-07:
if you check the issue tracker for the arduino-esp32 project you will see there are numerous issues with the i2c locking up under some circumstances requiring a reset to fix, it's possible that this library hits those conditions

olikraus, 2017-12-07:
anything what i could do in u8g2?

JamesGKent, 2017-12-08:
well there is an experimental branch thats getting good feedback, but from what i can gather (not tried it) it would require changes to the function calls used for i2c, and it's not included in master yet (they're still working on cleaning it up) so anything you do here would likely only work if people install the branch and will probably break again before it gets merged with master.
the big issue appears to be with calls to Wire.endTransmission(false); somehow locking up the i2c state machine, and then not being recoverable without a power cycle, i don't know if it's possible to avoid that?
also don't think thats the only issue.

olikraus, 2017-12-08:
good point. So you suggest u8g2 should use Wire.endTransmission(true)? (sending a stop condition instead of a restart)

tuupola, 2017-12-08:
> if you check the issue tracker for the arduino-esp32 project you will see there are numerous issues with the i2c locking up under some circumstances requiring a reset to fix, it's possible that this library hits those conditions

IIRC this was an issue with esp-idf itself and has now been fixed.

JamesGKent, 2017-12-08:
@tuupola i agree its an esp-idf issue, but it's far from solved:
[#741](https://github.com/espressif/arduino-esp32/issues/741)
[#839](https://github.com/espressif/arduino-esp32/issues/839)
[#349](https://github.com/espressif/arduino-esp32/issues/349)
[#246](https://github.com/espressif/arduino-esp32/issues/246)

and there are more, and as noted in #839 the current "fix" branch is not a drop in replacement

@olikraus that may solve the issue however i've only been loosely following the issue tracker, i have not played with i2c or the fixes posted so cannot really comment on how u8g2 could be adapted to work around the issues, however in general terms i'd say this is really something that needs to be fixed upstream in the esp32 project

olikraus, 2017-12-08:
ok, thanks for clarification. So for now, i leave u8g2 as it is. If there is any known workaround, please let me know.

closed 2017-12-30 by olikraus

ro269, 2018-02-02:
I am trying to interface a SSD1309 OLED with the ESP32 using I2C in the ESP-IDF environment. I am not able to get the screen working. Although the screen does appear in a I2C scan test. Is there an issue with ESP-IDF as mentioned in this thread or am I missing something? Please let me know.
Thanks.