olikraus/u8g2 #446: Very long lines of text may still wrap

MattMatic, 2017-12-04:
In testing the ProFont issues, I found that ProFont29 was producing an odd output. Looks like the line is so long that it's wrapping back and overwriting (I am drawing a white box around the text, then drawing the text with XOR mode).

The snapshot is the 160x104 display, with ProFont 10,11,15,17, and 22. The "EF" letters have been rendered over the "29". The "defABCD" are correctly hidden and not-rendered.

Trying to drill down and find where the issue is...
[image]

AnHardt, 2017-12-04:
The string for "22" is longer than 256 pixels.
Coordinates are 8 bit.
Values between 160 and 256 are clipped correctly.
Above 256 they appear as low numbers again -  8 bit overflow.

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159 ?

MattMatic, 2017-12-04:
Aha! Perfect. Thank you 👍

closed 2017-12-07 by olikraus