olikraus/u8g2 #447: Using u8g2 library with ssd1306 HW SPI on nRF52832

NailAgliev, 2017-12-05:
Hi. I want to using u8g2 library with OLED display controlled by ssd1306 connected to 4 wire HW SPI  interface on nRF52832 ARM controller.  I realized that I would need to create my own HW interface callback function. Do I also need to create a GPIO and Delay callback function to?

olikraus, 2017-12-05:
Yes

NailAgliev, 2017-12-05:
Thanks for the quick response. I also wanted to find out whether it is possible in any way to use the interrupt timer in the GPIO and Delay callback function. And in general, why do we need delays and maybe there is a way how to do without them?

olikraus, 2017-12-05:
Usually not much delay functionally is requied. It is mainly used for the reset of the display and for the SW emulated I2C and SPI transfer.

You find more details here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

NailAgliev, 2017-12-05:
I do not fully understand which messages I need to handle in GPIO and Delay callback function if I use HW SPI. Why I shoulduse u8x8_gpio_SetDC(u8x8, arg_int) to send a message to the GPIO procedure if i can set DC pin in HW interface callback function?

NailAgliev, 2017-12-09:
Спасибо за помощь я успешно смог запустить мой дисплей  nRF52832 используя HW SPI и вашу замечательную библиотеку. Как было сказано в #356 мне пришлось создать глобально буфер а также шрифт. У меня появился следующий вопрос. В каком формате шрифты используются в библиотеке и как я могу использовать другие шрифты которых нет в файле шрифтов?

olikraus, 2017-12-09:
You can use any font in bdf file format:

https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L179

There are also tools to create bdf fonts or convert ttf to bdf (fontforge, otf2bdf).

micooke, 2017-12-12:
@NailAgliev - can you put your port somewhere on github? That would be a great help!

olikraus, 2017-12-30:
no further feedback... closing...

closed 2017-12-30 by olikraus

NailAgliev, 2017-12-31:
@micooke Do you still need a code?

NailAgliev, 2018-01-01:
@micooke 
uint8_t u8x8_byte_nrf_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)		
{
	switch(msg)
		{
			case U8X8_MSG_BYTE_SEND:
				{
					APP_ERROR_CHECK(nrf_drv_spi_transfer(&spi, arg_ptr, arg_int, NULL, NULL));	
					break;
				}
			case U8X8_MSG_BYTE_INIT:
				{
					
					nrf_drv_spi_config_t spi_config = NRF_DRV_SPI_DEFAULT_CONFIG;
					spi_config.ss_pin   = SPI_SS_PIN;
					spi_config.miso_pin = SPI_MISO_PIN;
					spi_config.mosi_pin = SPI_MOSI_PIN;
					spi_config.sck_pin  = SPI_SCK_PIN;
					APP_ERROR_CHECK(nrf_drv_spi_init(&spi, &spi_config, NULL, NULL));
					
					nrf_gpio_cfg_output(DC_PIN);
					nrf_gpio_cfg_output(RESET_PIN);
					break;
				}
			case U8X8_MSG_BYTE_SET_DC:					//Pin Data/Comand
				{
					nrf_gpio_pin_write(DC_PIN, arg_int);
					break;
				}
		}
	return 1;
}

uint8_t u8g2_nrf_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) 
	{
	switch(msg) {
	// Initialize the GPIO and DELAY HAL functions.  If the pins for DC and RESET have been
	// specified then we define those pins as GPIO outputs.
		case U8X8_MSG_GPIO_AND_DELAY_INIT: 
			{
				nrf_gpio_cfg_output(RESET_PIN);
				nrf_gpio_cfg_input(SELECT_PIN, NRF_GPIO_PIN_PULLUP);
				nrf_gpio_cfg_input(NEXT_PIN, NRF_GPIO_PIN_PULLUP);
				nrf_gpio_cfg_input(PREV_PIN, NRF_GPIO_PIN_PULLUP);
				nrf_gpio_cfg_input(HOME_PIN, NRF_GPIO_PIN_PULLUP);
				break;
			}

	// Set the GPIO reset pin to the value passed in through arg_int.
		case U8X8_MSG_GPIO_RESET:
			{
				nrf_gpio_pin_write(RESET_PIN, arg_int);
				break;
			}
			
		case	U8X8_MSG_GPIO_MENU_SELECT:			
		{
			if(nrf_gpio_pin_read(SELECT_PIN)!= 1)
			{
				app_timer_stop_all();
				app_timer_start(my_timer, APP_TIMER_TICKS(WAIT_TIME), NULL);
				u8x8_SetGPIOResult(u8x8, 0);
				break;
			}
			break;
		}
		case	U8X8_MSG_GPIO_MENU_NEXT:
		{
			if(nrf_gpio_pin_read(NEXT_PIN)!= 1)
			{
				app_timer_stop_all();
				app_timer_start(my_timer, APP_TIMER_TICKS(WAIT_TIME), NULL);
				u8x8_SetGPIOResult(u8x8, 0);
				break;
			}
			break;
		}			
		case	U8X8_MSG_GPIO_MENU_PREV:
		{
			if(nrf_gpio_pin_read(PREV_PIN)!= 1)
			{
				app_timer_stop_all();
				app_timer_start(my_timer, APP_TIMER_TICKS(WAIT_TIME), NULL);
				u8x8_SetGPIOResult(u8x8, 0);
				break;
			}

			break;
		}			
		case	U8X8_MSG_GPIO_MENU_HOME:
		{
			if(nrf_gpio_pin_read(HOME_PIN)!= 1 || timer_flag == 1)
			{
				app_timer_start(my_timer, APP_TIMER_TICKS(10), NULL);
				u8x8_SetGPIOResult(u8x8, 0);
				return 1;
			}
			break;
		}			
		default:
     u8x8_SetGPIOResult(u8x8, 1);			// default return value
     break;
	}
	return 1; 

I hope this will be enough.

micooke, 2018-01-02:
@NailAgliev - thankyou for that, i look forward to trying it out :+1: