olikraus/u8g2 #450: SSD1309 - parallel mode not working (blank screen)

wdub, 2017-12-08:
I try to configure an 128x64 graphic display but unfortunately the screen remains black regardless of the constructor used(for the SSD1309 controller, of course).

Here's the code I tried:
`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

#define en 13 // enable
#define cs 11 // chip select
#define rs 10 // reset
#define dc 12 // D/C

int rsState = LOW;

U8G2_SSD1309_128X64_NONAME2_F_8080 u8g2(U8G2_R0, 2, 3, 4, 5, 6, 7, 8, 9, /*en*/en, /*cs*/cs, /*dc*/U8X8_PIN_NONE, /*rs*/rs);

void setup(void) {
  digitalWrite(rs, LOW);
  u8g2.begin();
}

void loop(void) {
  if (rsState == LOW) {
    digitalWrite(rs, HIGH);
    rsState = HIGH;
  }

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(0, 20, "Hello World!");
  u8g2.sendBuffer();
}`

and [specs of the display](https://www.tme.eu/en/Document/e7a4129fc86ed405d0506de1bf54cc9f/REX012864JYPP3N00000.pdf)

I also posted on arduino.stackexchange.com but no luck for me: https://arduino.stackexchange.com/questions/47420/graphic-display-through-8080-interface-on-arduino

Thank you

olikraus, 2017-12-08:
Is the display configured in 8080 mode?
Is wiring correct? Did you measure activity on the display pins?
Which Arduino Board did you use? 5V Boards require a level shifter from 5V to 3.3V.

wdub, 2017-12-08:
Thanks for the answer olikraus. Yes, the display is configured in 8080 mode (14 pins): VDD, VSS, D0-D7, CS, RES, DC and E.
R/W is connected to GND through a 2K resistor as I did with an old LCD.
I use Arduino Uno for this projects and voltage dividers were used for D0-D7, CS, RES, DC and E pins so no 5V coming out from those pins. Level shifters are on their way, i will receive them today.

A pic of the whole setup:
[image]

Maybe it's not relevant but I have to mention that I can see some pixels on the screen only if I remove the DC parameter from constructor.

I'll check the voltage on D0-D7 pins in a minute and get back to you with more info.

wdub, 2017-12-08:
Just checked and i can see 0.25V - 0.35V on D0-D4 pins and 2.3V-2.4V on D5-D7 pins.

olikraus, 2017-12-08:
> Yes, the display is configured in 8080 mode (14 pins): VDD, VSS, D0-D7, CS, RES, DC and E.

Usually there are some solder bridges in the board which make it work in 8080 or any other mode. So i mean it is not a question on how you did wired your display. The question is more what kind of interface is expected internally by the display.

wdub, 2017-12-08:
Oh I see. On the back of the PCB there are no jumpers, DIP switches or something to allow the switch to 8080 mode.
[image]
i2C and SPI are not supported by this model so I guess the parallel mode is set to default. 
Could the **E** pin be the culprit? Maybe it should be connected to GND through a resistor as I did with the R/W

olikraus, 2017-12-08:
It should be written in the datasheet of your display. Actually JB2 and JB1 look to me like the configuration solder bridges for the display. They are probably connected to BS1 and BS2 of the SSD1309. H probably means high and L low, so if you look into the SSD1309 datasheet you crosscheck, that this display is in 8080 mode (8080 mode is selected if BS1=high and BS2=high)

Did you connect the D/C pin to the dc pin of the constructor. At least this seems to be wrong in your first post.

Moreover, i think the E input must be pulled high, while the RW input of the display is the enable line on the constructor (for 8080 mode).

wdub, 2017-12-08:
Eureka! It works. Level shifters arrived and everything works like a charm. You were right regarding the E and R/W pins.
Here's the working code:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

//#define en 10
#define cs 11 // chip select
#define rs 13 // reset
#define dc 12 // D/C
#define en 10 // R/W on display - en on constructor

int rsState = LOW;

U8G2_SSD1309_128X64_NONAME2_F_8080 u8g2(U8G2_R0, 2, 3, 4, 5, 6, 7, 8, 9, /*en*/en, /*cs*/cs, /*dc*/dc, /*rs*/rs);

void setup(void) {  
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(0, 14, "Hello World!");
  u8g2.sendBuffer();
}
```
Last week I posted on reddit and I was told that the display may be fried because of the high voltage coming from data pins, I bought another one and now I have 2 working displays :).

Thank a lot for help and keep up the good work.
[image]

olikraus, 2017-12-08:
Great! Looks good.

One small note to readers of this thread: In U8g2 documentation "rs" is usually refered as "register selected" which is the same as d/c (data/command). In the above code only "rs" is refered as reset line.

closed 2017-12-09 by olikraus