olikraus/u8g2 #454: Struggeling implementing support for Waveshare 1.54 200x200 ePaper module

newkit, 2017-12-11:
Hey there,

I try to add support for the Waveshare 1.54 200x200 ePaper module.
I am guessing that would be pretty straight forward, since u8g2 is already supporting the 2.9 inch version.  
Unfortunately the datasheet of the 1.54 display does not state explicitly the controller (IL3820) as the 2.9 inch version does. What I tried so far is duplicating the code for the 2.9 inch version and adjusting the pixel dimensions for the 1.54 version, but without success so far.

Does anyone know, which controller is used in the 1.54 inch display?

olikraus, 2017-12-11:
What I understood from previous discussions is, that the constructor
U8G2_SSD1607_200X200_1_4W_HW_SPI
should work for this ePaper.
Did you try the same?

olikraus, 2017-12-30:
no feedback, closing...

newkit, 2017-12-31:
Sorry for my silence. Yeah, that constructor you provided works. Will check if the same LUT for fast refresh will work.

olikraus, 2017-12-31:
ok... thanks for the feedback

newkit, 2018-01-06:
I put this here just for reference:

I found the LUT in the latest Waveshare Example is slightly different from the one used for the `U8G2_SSD1607_200X200_1_4W_HW_SPI`: 

```
0x19, 0x01, 0x00
```

in the end. The entire LUTs:

```
const unsigned char lut_full_update[] =
{
    0x02, 0x02, 0x01, 0x11, 0x12, 0x12, 0x22, 0x22,
    0x66, 0x69, 0x69, 0x59, 0x58, 0x99, 0x99, 0x88,
    0x00, 0x00, 0x00, 0x00, 0xF8, 0xB4, 0x13, 0x51,
    0x35, 0x51, 0x51, 0x19, 0x01, 0x00
};

const unsigned char lut_partial_update[] =
{
    0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
    0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12,
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

I also noticed that the pixels near the border of the display get kind of "bleached" out - first they are okay, but within the update the become pale:

[image]

olikraus, 2018-01-06:
Looks like the "making black" Periode needs to be Longerich.

Xerxes3rd, 2018-01-07:
I have both the 2.9" version and the 1.54" version.  The U8G2_SSD1607_200X200_1_4W_HW_SPI constructor works fine for me as well on the 1.54" version, but the whole display on the 1.54" version seems lighter than the 2.9" version.

olikraus, 2018-01-07:
@Xerxes3rd 
Support for e-paper is very complicated. Without knowing the correct LUT for exactly your device support is almost impossible. U8G2_SSD1607_200X200 is optimized for the embedded adventures e-paper: https://www.embeddedadventures.com/epd_displays_category.html

Xerxes3rd, 2018-01-07:
The Waveshare display is a Good Display GDEP015OC1.  Their source found here: http://www.good-display.com/download_detail/downloadsId=518.html lists the full update LUT as:

`U8X8_A(0x50),
U8X8_A(0xAA),
U8X8_A(0x55),
U8X8_A(0xAA),
U8X8_A(0x11),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
	
U8X8_A(0xFF), /* Timing part of the LUT, 20 Phases with 4 bit each: 10 bytes */
U8X8_A(0xFF),
U8X8_A(0x1F),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
U8X8_A(0x00),
`

Using that LUT, I get a darker black color that's closer to the 2.9" version.

olikraus, 2018-01-07:
I have created a new constructor ...SSD1607_GD_200X200...
Can you test the same? The new constructor contains your LUT.

Xerxes3rd, 2018-01-07:
Looks good.

After I updated, I started having issues with the 296x128 (2.9 inch) version.  I had to add `u8g2.setFlipMode(0);` after `begin()` in order for that display to work correctly (apparently the 1.54 inch does not require `setFlipMode(0)`).  Apparently this changed since my previous version (2.19.8).  After I added that, everything works fine.

olikraus, 2018-01-08:
ToDo: Test  296x128 (2.9 inch) version and check whether u8g2.setFlipMode(0); is required.

Xerxes3rd, 2018-01-08:
Additional information: I'm running my 296x128 display in rotation mode 1 (`U8G2_R1`).  I'm not sure if the issue happens in default rotation (`U8G2_R0`).

Xerxes3rd, 2018-01-08:
Additional additional information: I re-tested this, and I think everything is OK (`setFlipMode(0)` not required).  In my previous tests, I think the Arduino IDE got confused when I switched out versions of U8g2.  If I move one version out and another in, the IDE doesn't always re-link everything.  Closing & re-opening the sketch seems to remove all the temporary files, which causes a fresh compilation.  It looks like everything's good.  Sorry for the confusion. =)

olikraus, 2018-01-08:
ok, great. Thanks for the feedback.

Jacotheron, 2018-01-17:
I have installed this to get my Waveshare 1.54" display working and it works. Thank you for the great work. One issue I was having is the hardware SPI does not want to be shared (using it with an SPI Bluetooth module), but I had 2 spare pins, so now running it on software SPI (well I got the display for not being fast, so that is not really an issue).

gnampf89, 2018-01-27:
I got the Waveshare 1.54 display to work with the Hello World example somehow, I had to add the delay loop from the original Waveshare code to see anything:
`while (digitalRead(BUSY_PIN) == HIGH) {
    delay(100);
  }`
After reset the first two loops show the last two frames before the reset, according to Waveshare documentation the display has two buffers. The 3rd and any subsequent updates show the frame generated in the previous loop.
The display is blinking twice on each loop (old frame -> black -> white -> black -> white -> new frame), it does not do that when using the Waveshare sample code.

glasseffects, 2018-01-28:
Hi Oliver, been using this library on OLEDs for quite some time. I recently got a Waveshare B/W 1.54 inch e-paper module and there seems to be an issue with refresh. As discussed above by gnampf89, the display cycles from black to white between frames. Is it possible to selectively refresh a part of display instead of the entire screen? Everything else looks brilliant on the screen!

Thanks.

olikraus, 2018-01-28:
> Is it possible to selectively refresh a part of display instead of the entire screen?

Unfortunately this is not possible. Actually I see a lot of differences between epaper and LCDs. U8g2 was planed for LCDs so some features for epaper are not part of the software architecture of u8g2.

Actually I think u8g2 supports GLCDs and OLEDs nicely, but u8g2 is not really designed for epaper.

glasseffects, 2018-01-29:
U8G2 is evidently the logical choice for OLEDs! As for e-paper displays, even though its not crafted for this purpose, it draws everything with immacualte perfection, the black and white display flickering and other glitches as disscussed by other members are the only signs of incompatibility.

Thanks nonetheless for all your efforts!

glasseffects, 2018-01-29:
Hi Oliver, can you suggest any alternative library for EPDs? I've tried GxEPD, and I don't quiet understand how the elements are drawn, I tried some Adafruit native codes for text and shapes, but it seems like there's some complex algorithm to display simple elements on the EPDs. After using this library, I've almost given up on EPDs.

olikraus, 2018-01-29:
I think @ZinggJM has created an excellent library here: https://github.com/ZinggJM/GxEPD

glasseffects, 2018-01-29:
Hi, this is precisely the library that has had me invest hours into drawing simple shapes from Adafruit Graphics Library. I tried to isolate the bare minimum working code from associated list of examples. The code below is known as "Partial_EPD_Update_Test" from GxEPD, here, all I want is to display some text, images and shapes and update them seperately. The issue is I'm not able to isolate the code for how its done with this library. This calls for an appreciation for the remarkable simplicity U8G2Lib offers over many other alternatives.

`
#include <GxEPD.h>
#include <GxGDEP015OC1/GxGDEP015OC1.cpp>    // 1.54" b/w

#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeMonoBold9pt7b.h>

#include GxEPD_BitmapExamples

// pins_arduino.h, e.g. AVR
#define PIN_SPI_SS    (10)
#define PIN_SPI_MOSI  (11)
#define PIN_SPI_MISO  (12)
#define PIN_SPI_SCK   (13)

GxIO_Class io(SPI, SS, 8, 9);
GxEPD_Class display(io);

void setup(void)
{
  Serial.begin(115200);
}

void loop()
{
  showPartialUpdate_AVR();
}

void showBlackBoxCallback(uint32_t v)
{
  uint16_t box_x = 10;
  uint16_t box_y = 15;
  uint16_t box_w = 70;
  uint16_t box_h = 20;
  display.fillRect(box_x, box_y, box_w, box_h, v);
}

void showValueBoxCallback(const uint32_t v)
{
  uint16_t box_x = 10;
  uint16_t box_y = 15;
  uint16_t box_w = 70;
  uint16_t box_h = 20;
  uint16_t cursor_y = box_y + box_h - 6;
  display.fillRect(box_x, box_y, box_w, box_h, GxEPD_WHITE);
  display.setCursor(box_x, cursor_y);
  display.print(v / 100);
  display.print(v % 100 > 9 ? "." : ".0");
  display.print(v % 100);
}

void showPartialUpdate_AVR()
{
  uint16_t box_x = 10;
  uint16_t box_y = 15;
  uint16_t box_w = 70;
  uint16_t box_h = 20;
  uint16_t cursor_y = box_y + box_h - 6;
  uint32_t value = 1395;
  display.setFont(&FreeMonoBold9pt7b);
  display.setTextColor(GxEPD_BLACK);
  display.setRotation(0);
  // draw background
  display.drawExampleBitmap(BitmapExample1, sizeof(BitmapExample1));
  delay(2000);

  // partial update to full screen to preset for partial update of box window
  // (this avoids strange background effects)
  display.drawExampleBitmap(BitmapExample1, sizeof(BitmapExample1), GxEPD::bm_flip_x | GxEPD::bm_partial_update);
  delay(1000);

  // show where the update box is
  for (uint16_t r = 0; r < 4; r++)
  {
    display.setRotation(r);
    display.drawPagedToWindow(showBlackBoxCallback, box_x, box_y, box_w, box_h, GxEPD_BLACK);
    //display.drawPagedToWindow(showBlackBoxCallback, box_x, box_y, box_w, box_h, GxEPD_BLACK);
    delay(1000);
    display.drawPagedToWindow(showBlackBoxCallback, box_x, box_y, box_w, box_h, GxEPD_WHITE);
  }
  // show updates in the update box
  for (uint16_t r = 0; r < 4; r++)
  {
    display.setRotation(r);
    for (uint16_t i = 1; i <= 10; i++)
    {
      uint32_t v = value * i;
      display.drawPagedToWindow(showValueBoxCallback, box_x, box_y, box_w, box_h, v);
      delay(500);
    }
    delay(1000);
    display.drawPagedToWindow(showBlackBoxCallback, box_x, box_y, box_w, box_h, GxEPD_WHITE);
  }
}
`

olikraus, 2018-02-13:
Not sure what else I can do here. Can we close this issue?

ZinggJM, 2018-02-14:
I have created new libraries GxEPD2_AVR and GxEPD2_32 on GitHub.com/ZinggJM.
These libraries are simpler to use than GxEPD and have similar interface to U8G2.
Still "under construction" and only black/white so far, but may be worth a try.

olikraus, 2018-02-14:
@ZinggJM  thanks for your great work on supporting the e-paper community. Initially i thought adding e-paper support to u8g2 would be a good idea, but I think your libs make a better job here.

For this ticket I had implemented GD support to the existing 200x200 driver, so i will close this ticket.

closed 2018-02-14 by olikraus

RonTheBestKingInTheWorld, 2018-09-06:
@newkit 
Did the partial update table from WS work for you?