olikraus/u8g2 #458: Enable 16bit mode without changing library

v1993, 2017-12-14:
I use this library in ESP-IDF project as git submodule (as I know, I can't share modifications) and want to use 16bit mode. How can I do it?

Adding

    CXXFLAGS+=-DU8G2_16BIT
    CFLAGS+=-DU8G2_16BIT
in `component.mk` does very fun (but wrong) stuff, like partially enabled 16bit mode.

P.S.: I include this library via `#include <U8g2lib.h>`.

olikraus, 2017-12-14:
Defining U8G2_16BIT in such a way is correct. Did you ensure that ALL .c files are recompiled?
Often you need to do a "make clean" first.

v1993, 2017-12-14:
I did `make clean`, still same problem.

olikraus, 2017-12-14:
Well, i can say this: I -DU8G2_16BIT works reliable. I do not know component.mk, instead i have my own, self written Makefile and i know that all .o files are deleted. After all .c files are recompiled, everything works with 16 bit.

I also do not know anything about ESP-IDF submodules. Maybe there is a mistake inside the integration of these submodules.

olikraus, 2017-12-30:
no further comments, closing...

closed 2017-12-30 by olikraus