olikraus/u8g2 #461: Storing a font on an SD card

RobotGB, 2017-12-21:
Hello,
After doing some testing, the only font that can fit on an Arduino's flash ROM, u8g2_font_b10_t_japanese1, is too simple for me, and has limited compatibility with some of the rarer characters that I would like to display; likewise, some characters which _can_ be displayed are too complex for the display to show accurately (for example, 戦勝 (Japanese for 'victory'), shows up as an unreadable block of pixels (not because of incompatibility in this case, but for the limitations that the font imposes).
Of course, it is not be possible to design a clearer font altogether, but is it possible to use a larger font and avoid storing it on the Arduino, instead on something like an SD card, then import it when in use?
Many thanks,
RobotGB

olikraus, 2017-12-21:
well... difficult... and maybe even beyond the scope of the project.

Let me consider the impact of this wish:
1. We need a SD-Card lib, like SD.h U8g2 will have a new dependency on SD.h. Some other users my already use a different SD-Lib (eg FatSD), so these users may end up with two libs. occuping twice the space and my have conflicts also. So might be forced to support several SD-Libs to satisfie users. I may need to introduce a none-SD version also, because otherwise people wo do not need SD card/feature may need SD.h. And there is one more case: People with Attiny my not have SD lib at all, so I have to consider this also

2. Access to glyphs will be slow. The page buffer rendering options in u8g2 in fact multiplies the access time. But if you have limited RAM (Arduino UNO) then you want to have page buffer rendering and of course you want the font on SD card. So: glyphs are accessed several times and access is very slow (SD card access).
Currently the font is a linked list of glphys. So U8g2 has to search the complete font data from beginning. It means, that I probably need a complete new font structure optimized for SD-Card access.
Because of the page buffer access, also chache mechanism is required so that we do not need to read from SD card all the time.

All in all, effort and risks are too high for this requires.

RobotGB, 2017-12-21:
I'll aim to find a workaround, but thanks for your advice!
I might be able to use bitmaps, possibly write it differently, with different parts of the glyph drawn as separate characters which can be reused, or make the glyphs clearer somehow.
Many thanks,
RobotGB

olikraus, 2017-12-21:
u8g2_font_b10_t_japanese1 is already white big.
Is there any smaller subset of japanese glyphs, which could be useful?

The current list of japanese unicode glyphs is here: https://github.com/olikraus/u8g2/blob/master/tools/font/build/japanese1.map

I derived this list from a probability count of the usage of each japanese gylph and took the 1000 most often used japanese unicode chars. Maybe we should cut this down to lesser unicode glyphs.

Is there a list of 500 glyphs which might be sufficient?

RobotGB, 2017-12-22:
My project requires using many glyphs, so this might not be suitable. One way to combat this might be to use a menu which displays the glyph upon request as a bitmap or so, and uses the standard font when not needed. Unfortunately, I won't be able to use my computer for a while, so I can't test it.
Since the file which I will be displaying values from (on an SD card) is already rather large, it might be useful not to store a larger font in memory.
Otherwise, I might be able to make my own design for the strokes in each character, using drawLine instead of bitmaps for the font. I'll see how it turns out.
Many thanks,
RobotGB

closed 2017-12-22 by RobotGB

olikraus, 2017-12-22:
All in all I suggest to use a uC with more flash memory. What about Arduino Zero?

RobotGB, 2017-12-22:
This suits me perfectly. Thanks for the rapid response! :smile:
Many thanks,
RobotGB