olikraus/u8g2 #462: Missing spaces after bdfconv

fire1, 2017-12-22:
This is not a bug in U8G2 library but... 
Somehow i destroy spaces in my custom font... i was using this command:
`bdfconv.exe -f 1 -m "32-1104,~403-899"  -n u8g_font_opel_ic_13 -o u8g_font_opel_ic_13.c OpelSansBold-13.bdf -v`

Any idea how to fix it  ?
Thanks!

olikraus, 2017-12-22:
I am not sure about the ~ operator.
Maybe try "32-403,899-1104".

fire1, 2017-12-22:
Ok in  "FontForge" space characters was marked as red, with opening and closing (without touching anything) edit option ... it FIXES this issue.
Thanks for time!

closed 2017-12-22 by fire1

olikraus, 2017-12-22:
:)

fire1, 2018-01-24:
P.S. Your command excludes more characters ... so ... it is better than my! (don't know way)

olikraus, 2018-01-24:
you mean "32-403,900-1104"? I think it is better to express what you want, than what you don't want ;-)