olikraus/u8g2 #463: Hardware SPI unstable

mrSynaps, 2017-12-22:
Hi,
I am using a Megatronics board with the ST7920 interface.  The software interface works just fine but the hardware interface is not stable. Is this a known issue? What can I do to prevent this. I really would like to use the hardware interface.

U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, 52); // not stable
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, 52, 51, 56); // works fine

olikraus, 2017-12-22:
ST7920 is very sensitive. How long are your wires? Did you solder the wiring?

olikraus, 2017-12-30:
no response, closing...

closed 2017-12-30 by olikraus

mrSynaps, 2017-12-30:
Uh yes we checked al that and that made it a little better. 
Can you give advice on another display which van be used in combination with hardware SPI, we really would like to free up the clock cycles
Switching to another display is at this stage no problem for us

olikraus, 2017-12-30:
ST7565 based Displays are much more reliable.