olikraus/u8g2 #464: sh1106

finopse, 2017-12-24:
hi master,
thank you for your library(u8g2), its work for my oled !

but, i have program used sh1106 library, and i cant make it to u8g2, can u help me? 

> > bool drawFrame2(SH1106 *display, SH1106UiState* state, int x, int y) {
>   display->setFont(ArialMT_Plain_24);
>   display->setTextAlignment(TEXT_ALIGN_LEFT);
>   display->drawString(32 + x, 0 + y, "PM 2.5");
>   display->drawString(10 + x, 32 + y, String(pm2_5) + "  ppm");
> }
> bool drawFrame3(SH1106 *display, SH1106UiState* state, int x, int y) {
>   display->setFont(ArialMT_Plain_24);
>   display->setTextAlignment(TEXT_ALIGN_LEFT);
>   display->drawString(30 + x, 0 + y, "Smoke");
>   display->drawString(10 + x, 32 + y, String(GAS_S) + "  ppm");
> }
> 
> bool drawFrame4(SH1106 *display, SH1106UiState* state, int x, int y) {
>   display->setFont(ArialMT_Plain_24);
>   display->setTextAlignment(TEXT_ALIGN_LEFT);
>   display->drawString(38 + x, 0 + y, "CO");
>   display->drawString(10 + x, 32 + y, String(CO) + "  ppm");
> }
> 

this is my program[](https://drive.google.com/file/d/1UgIEgr4t7n524rMPdBoiFeM3meJ3A1I0/view?usp=sharing)
and if used library sh1106, this is the images.h files [](https://drive.google.com/file/d/1gEyMxV8-CClkaPXofr5PKox4CXoC8f4J/view?usp=sharing)

thank you ^^

olikraus, 2017-12-24:
>  can u help me?
Maybe, if you tell me the problem you have with u8g2.

finopse, 2017-12-24:
my problem is why if i used lib sh1106 the oled will be like this 
[image]
(sorry its not my pic, but will be like that)
and if used u8g2 it will be okay.
but i have program with library sh1106.
do you know why?

thank you

olikraus, 2017-12-24:
Hmmm, I guess I can not help, because I do not know the sh1106 lib.

finopse, 2017-12-24:
yeah i hate sh1106, and now ill try to change the lib with u8g2, but i dont know u8g2 language,
if in sh1106 : 
> bool drawFrame2(SH1106 *display, SH1106UiState* state, int x, int y) {
>   display->setFont(ArialMT_Plain_24);
>   display->setTextAlignment(TEXT_ALIGN_LEFT);
>   display->drawString(32 + x, 0 + y, "PM [...]

just like i said before, i want change into the u8g2 language
 im sorry im very newbie about programing :(

big thank you

olikraus, 2017-12-24:
> i want change into the u8g2 language

Great, please read the u8g2 manual

> im sorry im very newbie about programing 

No problem, there are a lot of books and online tutorials on C, C++ and the Arduino Language.

finopse, 2017-12-25:
please can you give me example for my program

olikraus, 2017-12-25:
You mean? Examples for U8g2 are Part of the Library and can be found in the Examples section of the Arduino IDE

closed 2017-12-30 by olikraus