olikraus/u8g2 #465: u8g2.getStrWidth() strange behavior

pwiker, 2017-12-24:
When I use getStrWidth one time as follows:

Width = (int)(Screen.getStrWidth(TimeStr));

My return value is always 0.  However, if I call the function twice, one right after another:

Width = (int)(Screen.getStrWidth(TimeStr));
Width = (int)(Screen.getStrWidth(TimeStr));

My value comes back correctly.  I have verified that TimeStr is populated and the font has been set.

olikraus, 2017-12-24:
Maybe you need to activate 16 Bit Mode, See faq

pwiker, 2017-12-24:
Followed your advice with no change.  My screen is only 128x64 and the string is 53 pixels.  I really shouldn't be overflowing a byte.

olikraus, 2017-12-24:
Can you Post a complete examples?

pwiker, 2017-12-25:
After continued investigation, it does not look like getStrWidth() is at fault.  It seems that 
sprintf(TimeStr, "%02i:%02i", CurrentDateTime.Hour, CurrentDateTime.Minute);
corrupts TimeStr and hence gives problems to getStrWidth()

This, however, works just fine, measures correctly and whatnot.  This post was a false alarm. :-(
TimeStr[0] = '0';
TimeStr[1] = '7';
TimeStr[2] = ':';
TimeStr[3] = '0';
TimeStr[4] = '1';
TimeStr[5] = 0x00;

closed 2017-12-25 by pwiker

olikraus, 2017-12-25:
Ok...