olikraus/u8g2 #466: Thai font not displaying correctly

greanthai420, 2017-12-25:
Example
เทิด
will be shown as
เ ท  ิด

olikraus, 2017-12-27:
which font did you use?

olikraus, 2017-12-27:
The main problem is, that thai will require kerning information. However this kerning is not available to u8g2 font functions (it would also require a lot of space). However, there is a "secret" function which allows you to specify your own kerning.

This is the function:

```
u8g2.drawExtUTF8(u8g2_uint_t x, u8g2_uint_t y, uint8_t to_left, const uint16_t *kerning_table, const char *s)
```

It gets two more arguments compared to drawUTF8:
is_left: defines the direction of output
kerning_table: is your personal kerning table.

Let me take your example:
ทิด
actually only 
ทิ
is of interest here (right?). So ทิ is composed of ท and  ิ.
We need the unicode for this, so maybe use https://r12a.github.io/apps/conversion/

Unicode for ทิ are 0E17 0E34.

```
const uint16_t my_kerning_table[] = 
{
	/* first char, second char, gap reduction value */
	0x0E17, 0x0E34, 14, /* ท and  ิ */
  
        /* add more pairs here... */
	
	/* this line terminates the table */
	0xffff, 0xffff, 0xffff
};
```
The magic value is the 14. It depends on the glyph combination and the font size and has to be guessed or figured out manually.

For the font "u8g2_font_etl24thai_t", the following command will print ทิ correctly:

```
      u8g2.drawExtUTF8(&u8g2, 0, 20, 0, my_kerning_table, "ทิ");
```

Yet there is a bug in u8g2 with drawExtUTF8. Which needs to be fixed. You need to download a new beta release.... just a second.

olikraus, 2017-12-27:
You can download the beta release 2.21.3 here:

https://github.com/olikraus/U8g2_Arduino

Please install this beta release via the add zip library menu in your Arduino IDE.

olikraus, 2017-12-30:
no further feedback, closing...

Note: This issue fixes a problem with drawExtUTF8

closed 2017-12-30 by olikraus

greanthai420, 2018-01-12:
Sorry for the late reply, using custom kerning table and drawExtUTF8 works fine. Thank you.
How do i change the font though

olikraus, 2018-01-12:
> How do i change the font though

You could modify the font source, which is either .ttf or .bdf. Then apply bdfconv as mentioned in the u8g2 FAQ.