olikraus/u8g2 #467: Where is defined "u8x8_msg_cb" and "u8x8_t" in the C implementation

ihouses, 2017-12-26:
Hello,

First of all I am very new on this. I just want to use the u8g2 in a STM32 setup using System workbench for the ssd1306_128x64 using SPI and full buffer mode.

When I compile the code the compiler complains that can not resolve the data types"u8x8_msg_cb" and "u8x8_t" I was looking for the definition of such data typed but I can not find them either.

Does anyone know the root of the problem?

Thanks!

olikraus, 2017-12-27:
u8x8_t is defined here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L185

u8x8_msg_cb is defined here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L189

I have an example project for the stm32l0 and u8x8 here: https://github.com/olikraus/u8g2/tree/master/sys/arm/stm32l031x6/eval_board

Also have a look here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

olikraus, 2017-12-30:
no further feedback, closing...

closed 2017-12-30 by olikraus