olikraus/u8g2 #468: Particle Photon collapses

silvan2468, 2017-12-27:
Hy there
I have a Problem with the Microcontroller "Photon" from Particle.
If I use u8x8, "Hello World" is displayed with an SSD1327-Display.
But with u8g2, the microcontroller collapses allready at u8g2.begin(). First I thought there is not enough of RAM, but with "System.freeMemory();" I got, that when the u8x8-Hello world is running, I have everytime more than 60 kByte of free RAM. That should be more than enough. Also because the "Photon" (120 MHz, 128 kByte RAM) is much more powerful than every Arduino. But "Particle" tryed to make her system libraries very simmilar to Arduino.
Is that perhaps the problem? Do I have to change SPI-speed? How can I do this? And how is choosen SPI or SPI1?
Because the u8x8 works (more or less, I think there is something wrong with the coordinates; if I set the beginning of "Hello World" to (50/100) on a 128x128-display, it is more ore less centered. But with Y = 100 it should be on bottom...), I don't realy what the problem could be.
Page buffer mode I also didn't get working :(
Unfortunately, I'm also a beginner in C++.

Thanks a lot for help. 
Silvan

 #include <U8x8lib.h>
 #include <u8x8.h>
 #include <Arduino.h>
 #include <U8g2lib.h>
 #include <SPI.h>

// U8X8_SSD1327_MIDAS_128X128_4W_SW_SPI u8x8(/* clock=*/ D4, /* data=*/ D2, /* cs=*/ D5, /* dc=*/ D3, /* reset=*/ D6);                // u8x8 mode
U8G2_SSD1327_MIDAS_128X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ D4, /* data=*/ D2, /* cs=*/ D5, /* dc=*/ D3, /* reset=*/ D6);     // Full screen buffer mode
  //  U8G2_SSD1327_MIDAS_128X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ D4, /* data=*/ D2, /* cs=*/ D5, /* dc=*/ D3, /* reset=*/ D6);        // Page buffer mode

 void setup(void)
 {
  // Serial.begin(9600);

// u8x8 mode --------------------------------------------------------------------
   //u8x8.begin();

// Full screen buffer mode -----------------------------------------------------
   u8g2.begin();

// Page buffer mode ------------------------------------------------------------
  // pinMode(D5, OUTPUT);
  // pinMode(D3, OUTPUT);
  // digitalWrite(D5, 0);
  // digitalWrite(D3, 0);
  //  WiFi.off();
  // u8g2.begin();
  //  WiFi.on();
  // Particle.connect();

 }

 void loop(void) {
// u8x8 mode --------------------------------------------------------------------
  // u8x8.setFont(u8x8_font_chroma48medium8_r);
  // u8x8.drawString(50,100,"8x8-Mode!");
  // uint32_t freemem = System.freeMemory();
  // Serial.print("free memory: ");
  // Serial.println(freemem);
  // delay(1000);

// Full screen buffer mode -----------------------------------------------------
  //  u8g2.clearBuffer();					// clear the internal memory
  //  u8g2.setFont(u8g2_font_ncenB08_tr);
  //  u8g2.drawStr(50,100,"Full Screen");
  //  u8g2.sendBuffer();					// transfer internal memory to the display
  //  delay(1000);

// Page buffer mode ------------------------------------------------------------
  // u8g2.firstPage();
  // do {
  //   u8g2.setFont(u8g2_font_ncenB10_tr);
  //   u8g2.drawStr(0,50,"Hello");
  // } while ( u8g2.nextPage() );
  //  //delay(1000);

 }

olikraus, 2017-12-27:
Code looks correct. Yet, you could cleanup the include files. 
#include <u8x8.h> should not be required
#include <Arduino.h> should be the first include
#include <U8g2lib.h> is good for both u8x8 and u8g2 in Arduino Mode, so:
#include <U8x8lib.h> is not required.

summary: try this:
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

silvan2468, 2017-12-27:
Thanks olikruas for the fast answer.
But unfortunately this didn't help. And in my opinion the order of includes doesnt matter, because they are on top.
Some other ideas? 
And do you have paypal? If we will get it work, i realy like to donate for your work/help. For mechanics every one has to pay, but software has everytime to be for free...
But libraries save so much time.

olikraus, 2017-12-27:
> And in my opinion the order of includes doesnt matter, because they are on top.

Well, it should not matter, but unfortunately it depends on the person who wrote the include file: The include file must protect itself against double inclusion. While this is true for the u8g2 files, i do not know this about the other files.

> Some other ideas?

Can you verify that your system really hangs during begin()? Maybe you can make an LED blink after the statement. 

> Some other ideas?

Maybe your operating system does not init the u8g2 code correctly or the constructor is not correctly called.

> But libraries save so much time.

thanks

silvan2468, 2017-12-27:
The photon has a rgb led onboard. If the system crashes, it blinks sos in red. What it does, also when i uncomment all code expect the constructor and u8g2.begin(). I tried also with beginSimple() and uncommented power saving and clear display. It seams there is allready a problem with the initDisplay()...
Is it possible code does not work correct, also if compiling is ok?

silvan2468, 2017-12-28:
Correction...
I did it again with beginSimple() and set an LED when done. So, only beginSimple() works and the myc dont' crashes. Also when I call u8g2.setPowerSave(0), this works and the display shows immediatly a mix from dark an illuminated dots. But when I call u8g2.clearDisplay(), the myc crashes. 
So current state: it seems I have a Problem with "clearDisplay()".

olikraus, 2017-12-28:
Can you reproduce the crash also with other devices, like U8G2_SSD1306_NONAME_... Of course the display will not work, but the question is, whether the crash will happen.

silvan2468, 2017-12-29:
I tried it. But does also crash...
So i tried now to get the display running myself and was surprised successful. I am already able to set specific XY-dots. And by doeing myself, I know from every code line what it does.
But I will never get that high level of your library. So, if you have some further ideas how to solve my problem, please let me know.
lot of thanks

olikraus, 2017-12-29:
Strange...

olikraus, 2017-12-30:
another idea is to use complete different pins together with SW SPI:

Instead of 
// U8G2_SSD1327_MIDAS_128X128_1_4W_SW_SPI u8g2(U8G2_R0, / clock=/ D4, / data=/ D2, / cs=/ D5, / dc=/ D3, / reset=*/ D6); // Page buffer mode
use D7, D8, D9, D10, D11 (or any other pins, not matching any previous pins).#
Will it also crash?

silvan2468, 2017-12-30:
Doesn't matter. I have already tryed. And than, there would be already a problem with the initDisplay(). But it first crashes with clearDisplay(). InitDisplay seems to work. So SPI should work.

olikraus, 2017-12-30:
hmmm, then actually i do not know how to support further.
Can we close this issue? You could create a new issue or reopen the existing at any time.

silvan2468, 2017-12-30:
Pity :( but ok. I then will go further with my own library. Anyway, thanks a lot for your help!

Just a last question, please (not about this topic, but perhaps you had this problem, too):
If I draw just 8x8 letters or if I fill the whole display, every pixel has the same brightness. But if I write some 32x32-letters, the rows where the most pixels are, are slightly darker than the other pixels. Can you think why this happens? First I thought that's because of the most pixel and filling row after row, so the display can not lod the capacity of pixel correct. But if i fill the whole display, everything is ok...
That I don't write different grayscale pixels I have checked.
I know... Just curious questions from me ;)

silvan2468, 2017-12-30:
[image]
[image]

olikraus, 2017-12-30:
It looks like that the power supply is not good enough for your display. The display works like an old TV, where each line is written one by one. If a line contains all set pixel, then the overall current seems to be too much for the supply (pixel appear darker, compared to pixel rows with a lot of black).

You can try to reduce the max brightness or you my need to change the power supply for your display (better charge pump)

olikraus, 2017-12-31:
Regarding your crash: I did some tests with the gcc sanitizers, but I did not found any memory/pointer errors within u8g2. Without being able to reproduce the problem, I am not able to debug.

silvan2468, 2017-12-31:
Hmmm .. Strange. But thanks for testing.
I wish you a happy new year and good luck.

olikraus, 2018-01-01:
Happy New Year!
:-)

closed 2018-01-14 by olikraus

shonky2, 2019-03-09:
Hello @olikraus ,

As @silvan2468 said I also really appreciate all of your work on this library. I have been able to successfully use it on Arduino Mega and we are working on moving the software to a Particle Boron and I see this issue describes our same experience.

To simplify we started with your example code.

This is the code that I have been trying to flash:

#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <U8g2.h>

U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ D13, /* data=*/ D12, /* cs=*/ D14, /* dc=*/ D15);	// Enable U8G2_16BIT in u8g2.h
//U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Enable U8G2_16BIT in u8g2.h

#define DISP_CS         D14
#define SPI_DATA_CMD    D15
#define SPI_SCK         D13
#define SPI_MOSI        D12

void setup(void) {

  pinMode (DISP_CS , OUTPUT);
	
  pinMode (SPI_DATA_CMD , OUTPUT);
    
  digitalWrite(SPI_DATA_CMD , LOW);

  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
    delay(1000);  
    }

It is able to flash successfully but on start up on the Boron goes into a hard fault with single red error, which according to them means one of the following:

    Using an invalid pointer.
    Memory corruption caused by freeing memory twice, overwriting the end of a block of memory, etc.
    Making Wire (I2C) calls without calling Wire.begin().

I was curious to know if you ever came across a fix for this issue. Thank you!

olikraus, 2019-03-09:
Hmm... So the error means, that some wire procedures are called although SPI is used?

shonky2, 2019-03-09:
I'm not really sure if that is what that is saying but those are some of the generalized issues that could be occuring. Do you think that it is anything related to u8g2_cleardisplay.c? I saw that noted in issue #370 Looks like the failure is occurring in the u8g2.begin()

Tried commenting out, but still running into issue.
```
#include "u8g2.h"

/* Clear screen buffer & display reliable for all u8g2 displays. */
/* This is done with u8g2 picture loop, because we can not use the u8x8 function in all cases */
void u8g2_ClearDisplay(u8g2_t *u8g2)
{
  u8g2_FirstPage(u8g2);
  /* do {
  } while ( u8g2_NextPage(u8g2) );
  
    This function is usually called during startup (u8g2.begin()).
    However the user might want to use full buffer mode with clear and 
    send commands.
    This will not work because the current tile row is modified by the picture 
    loop above. To fix this, reset the tile row to 0, issue #370
    A workaround would be, that the user sets the current tile row to 0 manually.
  */
  u8g2_SetBufferCurrTileRow(u8g2, 0);  
}
```

Moved to HW SPI as well. Updated code below.

```
#include "Arduino.h"
#include "SPI.h"
#include "U8g2lib.h"

#define DISP_CS         D15
#define SPI_DATA_CMD    D16
//#define SPI_SCK         D13
//#define SPI_MOSI        D12

//U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, SPI_SCK, SPI_MOSI, DISP_CS, SPI_DATA_CMD);	// Enable U8G2_16BIT in u8g2.h
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, DISP_CS , SPI_DATA_CMD );	// Enable U8G2_16BIT in u8g2.h

void setup(void) {

  pinMode (DISP_CS , OUTPUT);
	
	pinMode (SPI_DATA_CMD , OUTPUT);
    
  digitalWrite(SPI_DATA_CMD , LOW);

  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```

I appreciate any help! Thank you!

olikraus, 2019-03-09:
I only have limited repo access and can not do any research at the moment. 
Another problem is this: U8G2 just calls wire.h and SPI.h procedures. This works fine with the original Arduino environment. Maybe the procedures for your board are broken.

shonky2, 2019-03-09:
Hi,

I've made some progress today. No more red light so it seems as thought the firmware is running. I've been running the following code and it is printing the serial lines up until 113 then stops. Also tried drawLine and that does the same thing. Never makes it to print line 117.

```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

#define clrBuf            u8g2.clearBuffer()
#define sndBuf                      u8g2.sendBuffer()

#define setBottom                u8g2.setFontPosBottom()
#define wrStr(x, y, b)            u8g2.drawStr(x, y, b)
#define wrLine(x, y, x1, y1)   u8g2.drawLine(x, y, x1, y1)

#define font14B                    u8g2.setFont(u8g2_font_helvB14_tr)

#define DISP_CS D15
#define SPI_DATA_CMD D16

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, DISP_CS, SPI_DATA_CMD); // Enable U8G2_16BIT in u8g2.h

void setup(){

Serial.begin(9600);

delay(1000);

pinMode(DISP_CS, OUTPUT);
pinMode(SPI_DATA_CMD, OUTPUT);

digitalWrite(SPI_DATA_CMD, LOW);

delay(2000);

Serial.println("starting u8g2");

u8g2.begin();

Serial.println("Line 90");

}

void loop(void){

Serial.println("Line 103");

clrBuf;

Serial.println("Line 109");

font14B;

Serial.println("Line 113");

wrStr(10,20,"This is a test");

Serial.println("Line 117");

sndBuf;

Serial.println("Line 121");

}
```

Any additional thoughts? Thank you for your help!

olikraus, 2019-03-10:
Line 113 and 117?

shonky2, 2019-03-10:
Those are the lines in the program. I just have those print so that I know how far through the program it gets before it hangs up. 
null

shonky2, 2019-03-11:
Hi @olikraus I feel like a fool. I think that I made a mistake in how I was defining pins. Will test later and let you know. Sorry for any inconvenience.

olikraus, 2019-03-11:
no problem....

shonky2, 2019-03-11:
Well, this is actually still a problem. The pins were labelled correctly. Still open to any thoughts if you can think of anything that might be causing this. Thank you!

olikraus, 2019-03-12:
The environment you use is unknown to me and the information available on the problem is still limited.

shonky2, 2019-03-12:
Alright, I appreciate your help. If I come up with any answers I'll definitely let you know.

suda, 2019-07-22:
@olikraus I tried to get you more details as this problem seems to relate to all the Particle devices (see #895). What I was able to gather was quite bizarre behavior. Namely that `u8g2` property of the `U8G2` class wasn't getting properly initialized.

When calling the `U8G2` constuctor, it calls `u8g2_Setup_FOO` (FOO being the controller/display string) by passing `u8g2` as a pointer:

```c
u8g2_Setup_uc1701_ea_dogs102_f(&u8g2, rotation, u8x8_byte_arduino_hw_spi, u8x8_gpio_and_delay_arduino);
u8x8_SetPin_4Wire_HW_SPI(getU8x8(), cs, dc, reset);
```
inside of the setup function the `u8g2` gets properly set up:

```c
u8g2_SetupDisplay(u8g2, u8x8_d_uc1701_ea_dogs102, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
buf = u8g2_m_13_8_f(&tile_buf_height);
u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
// At this point u8g2 has all the necessary properties
```

but the moment we're back in the setup function the pointer to `u8g2` points to an empty (not set up, the dimensions are zero and the `cb` is empty).

So later when calling  `u8g2.begin()` it calls `U8G2::clearDisplay()` which at some point calls `u8g2_SetBufferCurrTileRow()`:

```c
u8g2->tile_curr_row = row;
u8g2->cb->update_dimension(u8g2);
u8g2->cb->update_page_win(u8g2);
```

and the moment when trying to call `update_dimension()` it points to invalid memory causing the device to crash (and do the [SOS signal](https://docs.particle.io/tutorials/device-os/led/xenon/#red-flash-sos)).

Hope this helps and if you need more info, I'm happy to provide it :)

olikraus, 2019-07-22:
What kind of c-compiler do you use? Name & version?
I have a small idea:
Can you replace line 347 of u8g2.h:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L347
`#define u8g2_GetU8x8(u8g2) ((u8x8_t *)(u8g2))`
with
`#define u8g2_GetU8x8(u8g2) (&((u8g2)->u8x8))`

will this help?

olikraus, 2019-07-22:
... just want to mention, that you need to recompile everything. The above mentioned macro is one of the most often used macros in u8g2.

suda, 2019-07-23:
Unfortunately the `#define u8g2_GetU8x8(u8g2) (&((u8g2)->u8x8))` change still causes SOS.
We're using `gcc-arm-none-eabi` 5.3-2016q1

olikraus, 2019-07-23:
Hmmm... I have no idea what could be the reason.

axelinternet, 2020-03-16:
@suda @olikraus Did you guys get anywhere further regarding this? I can't figure out if anyone has managed to get u8g2 to work with Particle boards

olikraus, 2020-03-16:
No :( 
I don't own such board

jkunkee, 2021-11-17:
I ran into an interesting tidbit here: on my Particle Photon with DeviceOS 3.1.0 where I repro the hang, switching to [the C API](https://github.com/olikraus/u8g2/wiki/u8g2setupc) actually avoids the hang.

```
u8g2_Setup_ssd1327_i2c_ws_128x128_f(&u8g2, U8G2_R0, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
```

I wrote up my investigation [here](https://daskunkee.blogspot.com/2021/11/technical-ssd1327-oled-driver-on.html); TL;DR: I don't have a Particle debugger board, so I tried writing my own HAL to see what I could see. In doing so, I stumbled into this workaround. I have *no idea* why it works, alas!