olikraus/u8g2 #469: Two new/old China Displays

bobybobybob, 2017-12-27:
Hello Great Codemaster

do you know what Chip is used for that Mono 128x64 Orange Display?  https://de.aliexpress.com/item/LCD12864-LCD-Screen-128-64-Screen-128X64-LCD-Screen-Shield-Electrum-LED-backlight-Send-application-C51/32848043619.html

I have random test some and U8G2_ST7565_NHD_C12864_1_4W_SW_SPI looks very good

[image]

****************************************

I have no luck with HX1230 get random work with u8g2 

it has 96x68 Pixel   3.3V and is original maybe from nokia. 

Some other Codemaster made it work https://www.instructables.com/id/HX1230-LCD-Simple-Arduino-NANO-Clock/  

Maybe you can support that lcd with u8g2 too  or maybe it is already support but i dont know the right name in the big list.

olikraus, 2017-12-27:
The HX1230 looks like a PCF8812 controller:

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#pcf8812-96x65

bobybobybob, 2017-12-28:
I have tested some random

not work : U8G2_SSD1322_NHD_256X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Enable U8G2_16BIT in u8g2.h
not work : U8G2_SSD1327_SEEED_96X96_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);	// Seeedstudio Grove OLED 96x96
not work : U8G2_SSD1305_128X32_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_SSD1309_128X64_NONAME0_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
not work : U8G2_LD7032_60X32_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 11, /* data=*/ 12, /* cs=*/ 9, /* dc=*/ 10, /* reset=*/ 8);	// SW SPI Nano Board
not work : U8G2_UC1701_EA_DOGS102_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_UC1701_EA_DOGS102_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_PCD8544_84X48_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);						// Nokia 5110 Display
not work : U8G2_PCF8812_96X65_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Could be also PCF8814
not work : U8G2_PCF8812_96X65_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);						// Could be also PCF8814
not work : U8G2_KS0108_128X64_1 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 18, /*dc=*/ 17, /*cs0=*/ 14, /*cs1=*/ 15, /*cs2=*/ U8X8_PIN_NONE, /* reset=*/  U8X8_PIN_NONE); 	// Set R/W to low!
not work : U8G2_ST7565_64128N_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_ST7565_NHD_C12832_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
not work : U8G2_ST75256_JLX256128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Enable U8g2 16 bit mode for this display
not work : U8G2_NT7534_TG12864R_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
not work : U8G2_IST3020_ERC19264_1_6800 u8g2(U8G2_R0, 44, 43, 42, 41, 40, 39, 38, 37,  /*enable=*/ 28, /*cs=*/ 32, /*dc=*/ 30, /*reset=*/ 31); // Connect WR pin with GND
not work : U8G2_UC1638_160X128_1_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 2, /* dc=*/ 3, /* reset=*/ 4);  // Not tested 

Background light flicker U8G2_SSD1606_172X72_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);		// eInk/ePaper Display
Background light flickerU8G2_SSD1607_200X200_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// eInk/ePaper Display, original LUT from embedded artists

Mr. Chinese has release a chinese Dataguide but too much chinese i lever learned chinese language to understand it. Maybee you understand some of it https://cdn.instructables.com/ORIG/F4Y/Z9ZU/J76GKXJX/F4YZ9ZUJ76GKXJX.pdf

olikraus, 2017-12-28:
Did you Test U8G2_PCF8812_96X65_1_4W_SW_SPI?

bobybobybob, 2017-12-28:
Yes i tested all listed above and now again the U8G2_PCF8812_96X65_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Could be also PCF8814

The display has only CE and i have put CS and later DC on it with no reaction... ok little reaction.. the light flicker some rythmic. 

Backside is

GND | BL     | VCC   | CLK | DIN | N/C | CE | RST
GND| +3.3V| +3.3V|  13  |  11   |       |  9  |  8     No
GND| +3.3V| +3.3V|  13  |  11   |       | 10 |  8     Only Backlight Flicker

Shall i destroy my LCD and look behind the LCD so I maybe can find an IC Name or other usefull information for you? I have only one. Order a new one take time but ofcourse it is not very expensive, sometimes a little cheaper than the good old 5110nokia

olikraus, 2017-12-29:
well, i think there is too less of information, so adding support to u8g2 will be too difficult...

bobybobybob, 2017-12-29:
I have a clue :-) I add some Pictures i have made and that is all additional information i can give you. The Backside of LCD is only some Doubletape taped. I have cut it off to make new View. 

Maybe it is a litte usefull in future if that LCD get popular or nice Datapaper or help to identify the LCD
[image]
 panel. 
Then we close the topic and look forward to the future where maybe there is better data information for you and maybe it can be done if LCD gets ever little popular. I will contact you again if I can bring something interesting to new experience.

olikraus, 2017-12-30:
I have ordered one of these displays from here:
https://de.aliexpress.com/item/HX1230-LCD-screen-better-than-5110-resolution-96x68-picture-text-display/32810394997.html

It will usually take 4 weeks to reach me.

bobybobybob, 2017-12-31:
Dear Codemaster,

I feel very ashamed to have compelled you to buy this LCD. Excuse me. But also thanks for your help for me and other users.

olikraus, 2018-01-01:
No issue. It is (A) very cheap and (B) I did this before with many other displays. ;-)

olikraus, 2018-01-24:
display arrived... next step: implementation

54696d21, 2018-02-02:
*Politely asking* Hi, can you estimate how long this will take? (I really like the backlight of this display and want to use it for a project that I have to hand in mid-February.) No pressure at all (I have also other displays) :)

olikraus, 2018-02-02:
Well, yes... First I started to translate the chinese documentation. So here is the English version of the HX1230 command table:

Cmd | D/C | D7 | D6 | D5 | D4 | D3 | D2 | D1 | D0 | Description | Init Seq
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Set Power | 0 | 0 | 0 | 1 | 0 | W3 | W2 | W1 | W0 | W=1111: On, W=1000: Off | 0x02f
Set Contrast | 0 | 1 | 0 | 0 | B4 | B3 | B2 | B1 | B0 | Contrast? | 0x090
Set Invert Display | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | A0 | A0=0: Normal, A0=1: Inverse | 0x0a6
Set all Pixel On | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 | A0 | A0=0: Normal, A0=1: All On | 0x0a4
Display Enable | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 | A0 | A0=0: Off, A0=1: On | 0x0af
Set Page | 0 | 1 | 0 | 1 | 1 | 0 | Y2 | Y1 | Y0 | Page 0<=Y<=7 | -
Set Column Lo | 0 | 0 | 0 | 0 | 0 | X3 | X2 | X1 | X0 |   | -
Set Column Hi | 0 | 0 | 0 | 0 | 1 | 0 | X6 | X5 | X4 |   | -
Set Scan Start | 0 | 0 | 1 | S5 | S4 | S3 | S2 | S1 | S0 | Start Line 0<=S<=63 | 0x040
Write Data | 1 | D7 | D6 | D5 | D4 | D3 | D2 | D1 | D0 |   |

olikraus, 2018-02-02:
done

[image]

I personally would say, that there are better displays: The contrast is still a little bit high and can not be reduced further (maybe i should test 3.3V power supply). The seems to be only one backlight LED in the upper left corner which leads to a little bit unbalanced backlight. But no doubt, price is great.

olikraus, 2018-02-02:
please download U8g2 2.21.6 from here:
https://github.com/olikraus/U8g2_Arduino

Install via Arduino IDE (add zip lib menu). Will the HX1230 support work for you?

bobybobybob, 2018-02-11:
Thank you again, Codemaster
[image]

olikraus, 2018-02-12:
looks good :-)

54696d21, 2018-02-12:
Thank you very much :) It works for me too. Except for rotation levels R2 and R3 (as constructor value), so rotating by 180 and 270 degrees doesn't work at all.

olikraus, 2018-02-12:
Oh, then maybe I have to check the rotations. Thanks for the feedback.

olikraus, 2018-02-18:
> Thank you very much :) It works for me too. Except for rotation levels R2 and R3 (as constructor value), so rotating by 180 and 270 degrees doesn't work at all.

I checked this, for R2 and R3 some rows/columns are missing. i will do some further investigation.

olikraus, 2018-02-18:
Fixed R2 and R3.

closed 2018-02-18 by olikraus

bitbank2, 2018-05-01:
The commands to flip the HX1230 180 degrees are similar to other controllers: 0xc8 + 0xa1

olikraus, 2018-05-01:
Is this a question or a bug report?

bitbank2, 2018-05-01:
In your code repo, it appears that the flip180 logic for the HX1230 is commented out. I thought you would find it useful to know the 2 commands needed to implement it. (0xc0 & 0xa0 resets to normal).

olikraus, 2018-05-01:
oh, I did not know this... I will test this...

olikraus, 2018-05-01:
Excellent. Works without problem. Thanks a lot.

mcauser, 2019-08-17:
@olikraus 3 additional commands for your table above:

Cmd | D/C | D7 | D6 | D5 | D4 | D3 | D2 | D1 | D0 | Description | Init Seq
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Flip Horizontal | 0 | 1 | 0 | 1 | 0 | 0 | 0 | 0 | N/R | Normal 0xa0, Reverse 0xa1 | 0xa0
Flip Vertical | 0 | 1 | 1 | 0 | 0 | N/R | 0 | 0 | 0 | Normal 0xc0, Reverse 0xc8 | 0xc0
Software Reset | 0 | 1 | 1 | 1 | 0 | 0 | 0 | 1 | 0 | 0xe2 | -

If you flip both horizontal and vertical, it's effectively a rotate 180 deg.

If you don't use the RST pin (connect to VCC), you can perform a reset in software with 0xe2.

olikraus, 2019-08-17:
Great, thanks for the additional commands.