olikraus/u8g2 #471: I2c 128x32 Blue OLED LCD Display problem

karawin, 2017-12-31:
I am using u8g2 to support a lot of monochrome displays in a software for esp32 with esp_idf (not arduino).
using the HAL of https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2 
It works very well for many 128x64 ssd1306 and sh1106 and many other but i cannot find the right initialization for the 128x32 I2C Oled.
I tried 
u8g2_Setup_ssd1306_i2c_128x32_univision_f
u8g2_Setup_ssd1305_i2c_128x32_noname_f
without success.
With these inits a sh1106 in // shows only a line on the last line of pixels and the 128x32 oled is black.

help ;-)
Happy new year

olikraus, 2017-12-31:
for the univision device, try to modify the value in this line:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c#L51

Try 0x050 or 0x060 instead of 0x040.
Will this work?

karawin, 2017-12-31:
Thanks for your answer.
It is for a customer. I don't have this oled but I bough it and test it soon myself.
I will report the test next week in 2018.

olikraus, 2017-12-31:
you mean the ssd1306 128x32 shows one line with the sh1106 constructor?
Maybe you can post a picture.

karawin, 2017-12-31:
No, i only tested with the sh1106.
We will see that soon.
Thanks.

My project embed u8g2 and ucglib with HAL I made. Great library.
https://github.com/karawin/Ka-Radio32

olikraus, 2018-01-01:
nice project.

karawin, 2018-01-06:
Thanks.
I am back and with the right lcd now.
 the ssd1306 128x32  is black but the sh1106 on same i2c shows only a line on the last line.
Using u8g2_Setup_ssd1306_i2c_128x32_univision_f
With value 0x40 0x50 or 0x60. No change.

With u8g2_Setup_ssd1306_128x64_vcomh0_f the strange thing is that all lines are displayed but compressed.
See the picture below
[image]

olikraus, 2018-01-06:
There should be an alt0 constructor:
U8G2_SSD1306_128X64_ALT0_F
Can you try this?

karawin, 2018-01-06:
Not in c interface u8g2.h and for i2c.

olikraus, 2018-01-07:
> Missing a u8g2_Setup_ssd1306_i2c_128x32_alt0

Simply spoken: Your OLED is not supported by u8g2. The memory layout of your OLED differs from all known OLEDs.

I think best is to start with the univision constructor:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c

The alt0 setting should be correct already (bit 4 should be 0):
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c#L61

But maybe the start line and display offsets are different for your display:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c#L50

As stated initially, we need to try different values here. Somehow I came back to my initial answers.

karawin, 2018-01-07:
I think I will fed up.

Nothing is working with u8g2_Setup_ssd1306_i2c_128x32_univision_f
I tried to replace u8x8_d_ssd1306_128x32_univision_init_seq but the standard one is good.
I tried to replace u8x8_ssd1306_128x32_univision_display_info but without success.

The only way to have some display on it is with
u8g2_Setup_sh1106_128x64_noname_f
or
u8g2_Setup_ssd1306_128x64_vcomh0_f
or 
u8g2_Setup_ssd1306_128x64_noname_f
etc but nothing with the univision c interface.

olikraus, 2018-01-07:
Hmm    difficult to say something without having access to this board

karawin, 2018-01-10:
I made it working.
In fact i called the init with
u8g2_Setup_ssd1306_i2c_128x32_univision_f(
			&u8g2,
			U8G2_R0,
			u8g2_esp32_msg_i2c_cb,
			u8g2_esp32_msg_i2c_and_delay_cb); 

But the answer is
		u8g2_Setup_ssd1306_128x32_univision_f(
			&u8g2,
			U8G2_R0,
			u8g2_esp32_msg_i2c_cb,
			u8g2_esp32_msg_i2c_and_delay_cb); 

And this is true for all other types. If I call the i2c version of init, it fails.
May be because i am using a HAL interface?

olikraus, 2018-01-10:
could be, i have not written this HAL.

karawin, 2018-01-10:
It seems that https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2 is updated for this problem.
I close this issue.
Thanks for your support.

closed 2018-01-10 by karawin