olikraus/u8g2 #472: Using u8g2 library with ST7920 HW SPI on nRF52832

NailAgliev, 2018-01-04:
Hi again. I want to using u8g2 library with display controlled by ST7920 connected to 4 wire HW SPI interface on nRF52832 ARM controller.  I use the same code as in #447 but with inverse CS line capability. The same code works fine with ssd1306 but with ST7920 only slightly distinguishable individual pixels appear on the screen which are not at all similar to what I want to see. Maybe I'm missing something?

olikraus, 2018-01-04:
I did not follow the discussion much, but be aware on the following problems:

1. The SPI byte communication for the ST7920 differs from the SSD1306. U8g2 should be able to take care on this, still u8x8_byte_nrf_hw_spi has to communicate correctly with https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L324
2. ST7920 is a 5V device. ARM uC are ususally 3.3V devices, so it will not work without level shifting.

NailAgliev, 2018-01-05:
Thanks I managed to start the display and print the text but at the top of the display there is a strip of random pixels. 
[image]
All wires are soldered.

olikraus, 2018-01-05:
Hmm... Maybe the Display memory structure is different.

This means, some modification might be required for msg
U8X8_MSG_DISPLAY_DRAW_TILE
In fn
u8x8_d_st7920_common
Of file
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7920.c

olikraus, 2018-01-14:
no update, closing

closed 2018-01-14 by olikraus