olikraus/u8g2 #474: Use 128x32 pixel OLED like 16x2 LCD

You69Man, 2018-01-04:
I'd like to reengineer an existing project with a 16x2 LCD to a new minimized version using a 0.91" 128x32 pixel OLED display. I've tried the new u8x8 mode as the output is text only. On the 128x32 OLED display this allows 16x4 characters, which is basically great but not what I want, because all the display code (menu etc.) is designed for only two lines of text. 
So, I'm looking for a mode where I can use this display like it was a 16x2 display and in favor I would get a taller (better readable) fontsize. I'm thinking of a scaling mode for the existing 8x8 fonts so that they appear in double height. Is there already a solution or can it be made?

JamesGKent, 2018-01-04:
the 8x8 mode works by using character ROMs in the display modules themselves (i think) so there is no way to scale them, instead you could use u8g2 mode and select a monospaced font of the right height?

olikraus, 2018-01-04:
In principle you need a 8x16 font, where u8x8 only offers 8x8 by default.

I see two options. As mentioned from James, switching to u8g2 would be a good option. A suitable font might be pxplusibmvga8 (see https://github.com/olikraus/u8g2/wiki/fntgrpoldschoolpcfonts) which is almost like 8x16.

The other option would be to double the height of the u8x8 fonts. This would require some modification inside u8g2 library, but basically it would use the drawTile procedure to draw you own 8x8 tiles on the oled.
The content of the tile will be taken from the u8x8 fonts.
procedure will be this:

unsigned char buf[8];
read a glyph into buf. Use u8x8_get_glyph_data (https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_8x8.c#L62), but you need to remove the static keyword in the u8x8 code for this function.

extend this buffer:
buf[7] = buf[3];
buf[6] = buf[3];
buf[5] = buf[2];
buf[4] = buf[2];
buf[3] = buf[1];
buf[2] = buf[1];
buf[1] = buf[0];

use drawTile to draw buf at x,y

again read the same glyph into buf with u8x8_get_glyph_data

buf[0] = buf[4];
buf[1] = buf[4];
buf[2] = buf[5];
buf[3] = buf[5];
buf[4] = buf[6];
buf[5] = buf[6];
buf[6] = buf[7];

use drawTile to draw buf at x,y+1

You69Man, 2018-01-04:
Thanks for the quick answers! I'd like to avoid the u8g2 mode as I'm already short of RAM.
Thus the idea with the scaling of an existing 8x8 font in u8x8 mode. I thought maybe it's possible by the low-level functions of the g8x8 library, so that each glyph byte is automatically writte to two pixel addresses. But I haven't dug into the depth of the library yet. Maybe the proposal of olikraus already goes in this direction. So thanks for the while, I will do some research how to best solve my issue!

You69Man, 2018-01-04:
@olikraus: I've tried it like you have proposed but this only makes the glyphs wider but not taller. The glyphs seem to be rotated by 90°.

So I have extended u8x8_8x8.c now by following function (and additional made corresponding 1x2 string functions):
```
void u8x8_Draw1x2Glyph(u8x8_t *u8x8, uint8_t x, uint8_t y, uint8_t encoding)
{
  uint8_t i;
  uint16_t t;
  uint8_t buf[8];
  uint8_t buf1[8];
  uint8_t buf2[8];
  u8x8_get_glyph_data(u8x8, encoding, buf);
  for( i = 0; i < 8; i ++ )
  {
      t = u8x8_upscale_byte(buf[i]);
      buf1[i] = t >> 8;
      buf2[i] = t & 255;
  }
  u8x8_DrawTile(u8x8, x,   y, 1, buf2);
  u8x8_DrawTile(u8x8, x, y+1, 1, buf1);
}
```
It works great and also looks very good. So with this the very common 128x32 pixel displays (I2C), which are available for about 2€ in China can readily replace 16x2 LCD displays with only little software efforts.
So, I think 1x2 string versions could also be a nice future enhancement for the library.

You69Man, 2018-01-04:
Example with u8x8_font_pressstart2p_r
[image]

olikraus, 2018-01-04:
> but this only makes the glyphs wider but not taller.

Ah, yes true, the orientation is different. I forgot about my own data structures :-p

Excellent job. I will takeover this into u8x8.
Thanks for the picture.

You69Man, 2018-01-04:
You're welcome! Also thanks again for the great library!

olikraus, 2018-02-13:
this is added and tested... so closing this issue...

closed 2018-02-13 by olikraus