olikraus/u8g2 #475: Question re pinout for t6963c display in plain c

jpunkt, 2018-01-04:
hello,

i am trying to use a display with the t6963c controller on an atmega328p with plain c, using the parallel interface. being a noob, i am having some trouble with the "uC specific" callback function, as the datasheet uses different abbreviations for the pins, and that confuses me.

the datasheet gives the following description of the interface:

```
Pin     Name        Description

1       FG          Frame ground
2       VSS         Ground
3       VDD         Supply voltage for logic
4       VO          Input voltage for LCD
5       WR          Write signal
6       RD          Read signal
7       CE          Chip enable signal
8       C/D         H : Instruction Code, L : Data
9       NC          No connection
10      RST         Reset signal
11-18   DB0-DB7     Data bits 0 to 7
19      FS          Font select signal ( H : 6 x 8 dots, L : 8 x 8 dots )
20      VOUT        Output voltage for LCD
21      A           EL / LED Backlight anode
22      K           EL / LED Backlight cathode
```

I have attached the pins of my uC to the display like so:

```
uC      Display
PortD   DB0-DB7
PB0     WR
PB1     RD
PB3     CE
PB2     CD
PB4     RST
PB5     FS
```

but i don't know how this translates to the callback function. below is my implementation of the callback function, and the "easyavr.h" library i'm using to set the pins. any help is very much appreciated!

thank you,
jpunkt

[easyavr_h.txt](https://github.com/olikraus/u8g2/files/1603813/easyavr_h.txt)
[u8g2callback_c.txt](https://github.com/olikraus/u8g2/files/1603815/u8g2callback_c.txt)
[u8g2callback_h.txt](https://github.com/olikraus/u8g2/files/1603816/u8g2callback_h.txt)

olikraus, 2018-01-04:
at least partly this is answered in the FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L82

summery: Hardwire FS to GND, hardwire RD to 5V, so PB1 and PB5 are not required.

Msg U8X8_MSG_DELAY_I2C can be empty because it is only used for I2C interface. But it is also wrong, because the delay function will not accept float values (you need to round up to next integer, but as said before, it is not required at all)

Everything else should be fine.

jpunkt, 2018-01-05:
Thank you very much for your quick reply! somehow i missed the docs/faq.. things are a lot clearer now.

closed 2018-01-05 by jpunkt

olikraus, 2018-01-05:
:)