olikraus/u8g2 #476: how to display chinese font with u8g2 API

wxiangcheng, 2018-01-07:
I can display string "hi" on my OLED screen. How to display chinese font "你好" with u8g2 API in ESP32 (not Arduino)? I can't find the sample code but I guess i'm near the target. Please help.
My sample code:
`u8g2_SetFont(&u8g2, u8g2_font_unifont_t_chinese1);
	ESP_LOGI(TAG, "u8g2_DrawStr");
    u8g2_DrawStr(&u8g2, 2,17,"hi");
	u8g2_DrawGlyph(&u8g2, 8,17,"好");
	u8g2_DrawUTF8(&u8g2, 4,17,"你");`

olikraus, 2018-01-07:
Which constructor did you use?
Did you Fall sendBuffer?

wxiangcheng, 2018-01-08:
thanks olikraus. It may be not good to ask an application question here. I'm just wondering how to display chinese font easily with u8g2 API - sample code will be good. And now I could use  u8g2_DrawGlyph to display it with u8g2_font_unifont_t_chinese1 but the LIB is too limitted in serveral hundreds.. And It seems I have to use some tool to convert my TEXTs into new font_chinese_new... Sorry I'm for the first time to use u8g2 and I should read more document. Any comments are appreciated.

olikraus, 2018-01-08:
U8g2 is mainly intended for Arduino Users. It comes with chinese examples. They can be found in the examples menu of the Arduino IDE. 
It still does not work for your? I guess i need to know more details on your environment.

wxiangcheng, 2018-01-10:
I could display chinese font on my OLED with default u8g2_font_unifont_t_chinese3.
but I must use API "u8g2_DrawGlyph(&u8g2, 8,17,0x4e50)". 0x4e50 means chinese text "乐". Is there any common API that I can use directly? LIke "u8g2_Draw_xxxxx(&u8g2, 8,17,"乐")?

olikraus, 2018-01-10:
```
u8g2_DrawUTF8(&u8g2, 8,17,"乐");
```
should work. However there are some preconditions, which are not under control of u8g2: Your editor and your compiler need to work in UTF8 mode. Especially the editor is often a problem.

wxiangcheng, 2018-01-11:
thanks Olikraus. which editor you use? I tied VI, ultraedit - neither works. :(

wxiangcheng, 2018-01-11:
I find the setting of UTF8 in ultraedit. it is in advanced setting. Thanks again.

olikraus, 2018-01-11:
I use scite editor and indeed, the UTF8 options are often hidden.

closed 2018-01-14 by olikraus