olikraus/u8g2 #479: HW vs SW on Teensy 3.5, HW doesn't work

fmulassano, 2018-01-12:
I want to try the spi hardware version but it doesn't works for me..
i'm on teensy 3.5.
Works
U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 20, /* data=*/ 28, /* cs=*/ 27, /* dc=*/ 31, /* reset=*/ 8); // Enable U8G2_16BIT in u8g2.h

don't work.. 
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ 27,/* dc=*/ 31, /* reset=*/ 8); // Enable U8G2_16BIT in u8g2.h

any advice?
Thanks

olikraus, 2018-01-12:
HW_SPI will use SPI.h and the default SPI pins of your board, which (according to https://www.pjrc.com/teensy/td_libs_SPI.html) seems to be pins 13 and 11 instead of 20 and 28.

fmulassano, 2018-01-12:
thanks, i try

olikraus, 2018-01-18:
no further feedback... closing...

closed 2018-01-18 by olikraus

fmulassano, 2018-01-18:
Sorry, finally with the right pins, it works
Than you!

olikraus, 2018-01-18:
:)