olikraus/u8g2 #481: Support for HX8705-B controllers

Crazyphil, 2018-01-13:
There are some E-Ink displays with high DPI resolutions, fast refresh rates and grayscale support using a Himax HX8705-B controller (Data sheet: https://www.smart-prototyping.com/image/data/9_Modules/EinkDisplay/GDE043A2/HX8705-B_DS_preliminary_v01_100621.pdf).
One example is the 4.3" 800x600 screen from Good Display (Product page: http://www.buy-lcd.com/index.php?route=product/product&product_id=12515).

Unfortunately, I'm not very experienced in the display area. Is this a controller for which support could be added to u8g2?

olikraus, 2018-01-13:
This is a controller without display buffer. Very very difficult to drive this e-paper...

olikraus, 2018-01-13:
As a consequence of the missing display buffer on the controller, you need the buffer in your uC. Because of the e-paper technology, the buffer is required twice. For the 4 graylevels, it would require 800x600x2x2/8 = 240KBytes RAM. I do not know any Arduino Board which has this amount of RAM.

Crazyphil, 2018-01-15:
So the controller really just is there to directly display any data that is pumped into it on the EPD, which I assumed when reading the data sheet?

The double buffer is needed so the display can be cleared without long-term effects using a device-specific LUT, which I therefore assume also isn't done by the controller. If I streamed the current display data to the µC (I have a wireless door display in mind), I would need just a bit more than 120 KB RAM (the old display data + the network receive buffer) for this task.

Apart from that, there are Arduinos with enough RAM, namely the 32-bit variants Tian and Industrial 101, each having luxurious 64 MB (or, alternatively, the famous EPS32, which isn't technically an Arduino but programmable using the Arduino platform).

JamesGKent, 2018-01-15:
@Crazyphil the Tians MIPS core has 64MB of ram, but the arm core that you can program with arduino only has 32KB, likewise the industrial 101 has a MIPS core running linux with 64MB of ram, but the programmable arduino core only has 2.5KB so @olikraus is correct in saying there aren't any arduino cores with the required memory.

olikraus, 2018-01-18:
Well then, I hope it is fine to close this topic. U8g2 is designed for monochrome displays with internal memory, which is not true for the requested controller.

closed 2018-01-18 by olikraus