olikraus/u8g2 #484: New to OLEDs

MSKeith, 2018-01-15:
Hello,
    I am sure this is a dumb question, but I am using a SH1106 I2C OLED screen, and I am trying to figure out the pin lay outs for the SCL and SDA.  The constructor that I am using is:

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Figured that out by other post.  But when I am trying to understand what is the pin lay out that I need to enter...I am a bit confused.

I am using an ESP8266 board and have the SCL and SDA pins connected to the ESP8266 pins D1(GPIO 5 and D2(GPIO 4) respectfully.  

Can someone please explain to me how to place my pin out configuration into the constructor line?

Thanks

MSKeith, 2018-01-15:
Ok... so I figured out that this is rotation (U8G2_R0)... but how do I tell u8g2 to use my output pins for SCL and SDA?

olikraus, 2018-01-15:
I start with refering to the FAQ:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L35

Question: Is the answer in the FAQ good enough or does it need further explaination?

MSKeith, 2018-01-15:
Thank you for the response.  The link helped a lot.  But after reading FAQ...I noticed that pin remapping can be done...only with the esp8266...which is what I am using.  How would I set my pin configuration?

MSKeith, 2018-01-15:
Looking at this...would the constructor look like this?

U8G2_SH1106_128X64_NONAME_1_HW_I2C(rotation, [reset [, scl, sda])

olikraus, 2018-01-15:
yes. But it is a board specific extension, so I am not 100% sure whether this will work.

MSKeith, 2018-01-24:
So this is the exact constructor that I am using...

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0/*Rotation*/, /* reset=*/ U8X8_PIN_NONE);

so in review... should this constructor look like:

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0/*Rotation*/, /* reset=*/ U8X8_PIN_NONE, scl, sda);

Or better for my pinout:

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0/*Rotation*/, /* reset=*/ U8X8_PIN_NONE, /*scl*/5, /*sda*/4);

As I am connected on D1 and D2 for ESP8266.

MSKeith, 2018-01-25:
Hmmm... I am having some difficulties with this... Not sure why.  The sketch that I am using for my testing is pretty simple:

`/*Hello Wolrd.ino*/
#include <Arduino.h>

#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI

#include <SPI.h>

#endif
#ifdef U8X8_HAVE_HW_I2C

#include <Wire.h>
#endif
void setup(void) {

	u8g2.begin();

}

void loop(void) {

	u8g2.clearBuffer();					// clear the internal memory

	u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font

	u8g2.drawStr(0, 10, "Hello World!");	// write something to the internal memory

	u8g2.sendBuffer();					// transfer internal memory to the display

	delay(1000);

}
`
I have the board configured for d1 and d2 on the ESP8266 board to connect to scl and sda.  According to the pinout for the board d1 and d2 should be GPIO 5 and 4.

When I compile and upload the sketch to the ESP8266 I am getting no display on the screen.

The only think that I can think is wrong is the constructor line...but I am not sure how to verify this...has anyone run into this before?

Thank you in advance.

olikraus, 2018-01-25:
Somehow the code tags are wrong in the above code. I also miss the constructor line itself.

MSKeith, 2018-01-25:
Interesting... Let me try that again...

/*Hello Wolrd.ino*/
#include <Arduino.h>

#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI

#include <SPI.h>

#endif
#ifdef U8X8_HAVE_HW_I2C

#include <Wire.h>
#endif

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0/*Rotation*/, /* reset=*/ U8X8_PIN_NONE,5, 4);

void setup(void) {

	u8g2.begin();

}

void loop(void) {

	u8g2.clearBuffer();					// clear the internal memory

	u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font

	u8g2.drawStr(0, 10, "Hello World!");	// write something to the internal memory

	u8g2.sendBuffer();					// transfer internal memory to the display

	delay(1000);

}

So for what ever reason.. the code insertion tags are not allowing some parts to come in... Here is without the code tags...

olikraus, 2018-02-13:
hmm... still no success?

closed 2018-03-24 by olikraus