olikraus/u8g2 #486: How to correctly add your font to U8G2?

phoenix7575, 2018-01-19:
In the library there are very few Russian fonts there is only 3. On this page there are fonts of different sizes http://arduino.ru/forum/programmirovanie/russkie-shrifty-dlya-biblioteki-u8glib  .
Please tell me how to convert them for the library U8G2?
Need for project https://github.com/karawin/Ka-Radio32  .
The main tell me how to convert the specified fonts I will not understand, I lack experience. Please, help.
How to convert this font to bdf ?  Or how to use this font in U8G2.
/*
  Fontname: rus4x6  (Cyrillic 4x6 LCD Font)
  Copyright: (C) 2015	  Made by Maxim Z.	E-mail: <a href="mailto:Maxim_EV@bk.ru">Maxim_EV@bk.ru</a>
  Capital A Height: 5, '1' Height: 5
  Calculated Max Values w= 4 h= 6 x= 3 y= 3 dx= 4 dy= 0 ascent= 5 len= 6
  Font Bounding box     w= 4 h= 6 x= 0 y=-1
  Calculated Min Values           x= 0 y=-1 dx= 0 dy= 0
  Pure Font   ascent = 5 descent=-1
  X Font      ascent = 5 descent=-1
  Max Font    ascent = 5 descent=-1
*/

const u8g_fntpgm_uint8_t rus4x6[1768] U8G_FONT_SECTION(".progmem.rus4x6") = {
  0,4,6,0,255,5,1,102,2,189,32,255,255,5,255,5,
  255,0,0,0,4,3,0,1,5,5,4,1,0,128,128,128,
  0,128,3,2,2,4,0,3,160,160,4,5,5,4,0,0,
  160,240,160,240,160,3,6,6,4,0,255,64,224,192,32,224,
  64,3,5,5,4,0,0,128,32,64,128,32,4,5,5,4,
  0,0,64,160,64,160,80,1,2,2,4,1,3,128,128,2,
  6,6,4,1,255,64,128,128,128,128,64,2,6,6,4,0,
  255,128,64,64,64,64,128,3,5,5,4,0,0,160,64,224,
  64,160,3,5,5,4,0,0,64,64,224,64,64,2,2,2,
  4,0,255,64,128,3,1,1,4,0,2,224,1,1,1,4,
  1,0,128,3,5,5,4,0,0,32,32,64,128,128,3,5,
  5,4,0,0,64,160,224,160,64,3,5,5,4,0,0,64,
  192,64,64,224,3,5,5,4,0,0,64,160,32,64,224,3,
  5,5,4,0,0,224,32,64,32,192,3,5,5,4,0,0,
  160,160,224,32,32,3,5,5,4,0,0,224,128,192,32,192,
  3,5,5,4,0,0,96,128,192,160,64,3,5,5,4,0,
  0,224,32,64,128,128,3,5,5,4,0,0,96,160,64,160,
  192,3,5,5,4,0,0,64,160,96,32,192,1,4,4,4,
  1,0,128,0,0,128,2,5,5,4,0,255,64,0,0,64,
  128,3,5,5,4,0,0,32,64,128,64,32,3,3,3,4,
  0,1,224,0,224,3,5,5,4,0,0,128,64,32,64,128,
  3,5,5,4,0,0,192,32,64,0,64,3,5,5,4,0,
  0,96,160,160,128,96,3,5,5,4,0,0,64,160,224,160,
  160,3,5,5,4,0,0,192,160,192,160,192,3,5,5,4,
  0,0,64,160,128,160,64,3,5,5,4,0,0,192,160,160,
  160,192,3,5,5,4,0,0,224,128,192,128,224,3,5,5,
  4,0,0,224,128,192,128,128,3,5,5,4,0,0,96,128,
  160,160,96,3,5,5,4,0,0,160,160,224,160,160,3,5,
  5,4,0,0,224,64,64,64,224,3,5,5,4,0,0,32,
  32,32,160,64,3,5,5,4,0,0,160,160,192,160,160,3,
  5,5,4,0,0,128,128,128,128,224,3,5,5,4,0,0,
  160,224,224,160,160,3,5,5,4,0,0,32,160,224,160,128,
  3,5,5,4,0,0,64,160,160,160,64,3,5,5,4,0,
  0,192,160,192,128,128,3,6,6,4,0,255,64,160,160,160,
  64,32,3,5,5,4,0,0,192,160,192,160,160,3,5,5,
  4,0,0,96,128,64,32,192,3,5,5,4,0,0,224,64,
  64,64,64,3,5,5,4,0,0,160,160,160,160,224,3,5,
  5,4,0,0,160,160,160,224,64,3,5,5,4,0,0,160,
  160,224,224,160,3,5,5,4,0,0,160,160,64,160,160,3,
  5,5,4,0,0,160,160,64,64,64,3,5,5,4,0,0,
  224,32,64,128,224,2,5,5,4,1,0,192,128,128,128,192,
  3,5,5,4,0,0,128,128,64,32,32,2,5,5,4,0,
  0,192,64,64,64,192,3,2,2,4,0,3,64,160,3,1,
  1,4,0,255,224,2,2,2,4,1,3,128,64,3,4,4,
  4,0,0,96,160,160,96,3,5,5,4,0,0,128,192,160,
  160,192,3,4,4,4,0,0,96,128,128,96,3,5,5,4,
  0,0,32,96,160,160,96,3,4,4,4,0,0,64,160,192,
  96,3,5,5,4,0,0,32,64,224,64,64,3,5,5,4,
  0,255,96,160,96,32,192,3,5,5,4,0,0,128,192,160,
  160,160,3,5,5,4,0,0,64,0,192,64,224,3,6,6,
  4,0,255,32,0,32,32,32,192,3,5,5,4,0,0,128,
  160,192,160,160,3,5,5,4,0,0,192,64,64,64,224,3,
  4,4,4,0,0,160,224,160,160,3,4,4,4,0,0,192,
  160,160,160,3,4,4,4,0,0,64,160,160,64,3,5,5,
  4,0,255,192,160,192,128,128,3,5,5,4,0,255,96,160,
  160,96,32,3,4,4,4,0,0,160,192,128,128,3,4,4,
  4,0,0,96,192,32,192,3,5,5,4,0,0,64,224,64,
  64,32,3,4,4,4,0,0,160,160,160,96,3,4,4,4,
  0,0,160,160,160,64,3,4,4,4,0,0,160,160,224,160,
  3,4,4,4,0,0,160,64,64,160,3,5,5,4,0,255,
  160,160,96,32,192,3,4,4,4,0,0,224,32,64,224,3,
  6,6,4,0,255,32,64,192,64,64,32,1,5,5,4,1,
  0,128,128,128,128,128,3,6,6,4,0,255,128,64,96,64,
  64,128,4,2,2,4,0,3,80,160,255,3,5,5,4,0,
  255,192,160,192,128,128,3,4,4,4,0,0,96,128,128,96,
  3,4,4,4,0,0,224,64,64,64,3,5,5,4,0,255,
  160,160,96,32,192,3,6,6,4,0,255,64,64,160,160,64,
  64,3,4,4,4,0,0,160,64,64,160,3,5,5,4,0,
  255,160,160,160,224,32,3,4,4,4,0,0,160,160,96,32,
  3,4,4,4,0,0,160,224,224,224,4,5,5,4,0,255,
  160,224,224,240,16,4,4,4,4,0,0,192,96,80,96,4,
  4,4,4,0,0,144,208,176,208,3,4,4,4,0,0,128,
  192,160,192,3,4,4,4,0,0,192,96,32,192,4,4,4,
  4,0,0,160,208,208,160,3,4,4,4,0,0,96,160,96,
  160,3,5,5,4,0,0,64,160,224,160,160,3,5,5,4,
  0,0,224,128,192,160,192,3,5,5,4,0,0,192,160,192,
  160,192,3,5,5,4,0,0,224,128,128,128,128,3,6,6,
  4,0,255,224,96,160,160,224,160,3,5,5,4,0,0,224,
  128,192,128,224,3,5,5,4,0,0,160,224,64,224,160,3,
  5,5,4,0,0,192,32,64,32,192,3,5,5,4,0,0,
  128,160,224,160,32,3,6,6,4,0,255,96,128,160,224,160,
  32,3,5,5,4,0,0,160,160,192,160,160,3,5,5,4,
  0,0,96,160,160,160,160,3,5,5,4,0,0,160,224,160,
  160,160,3,5,5,4,0,0,160,160,224,160,160,3,5,5,
  4,0,0,64,160,160,160,64,3,5,5,4,0,0,224,160,
  160,160,160,3,5,5,4,0,0,192,160,192,128,128,3,5,
  5,4,0,0,64,160,128,160,64,3,5,5,4,0,0,224,
  64,64,64,64,3,6,6,4,0,255,160,160,160,64,64,128,
  3,5,5,4,0,0,64,224,160,224,64,3,5,5,4,0,
  0,160,160,64,160,160,3,5,5,4,0,0,160,160,160,160,
  224,3,5,5,4,0,0,160,160,96,32,32,3,5,5,4,
  0,0,160,160,224,224,224,4,6,6,4,0,255,160,160,224,
  224,240,16,4,5,5,4,0,0,192,64,96,80,96,4,5,
  5,4,0,0,144,144,208,176,208,3,5,5,4,0,0,128,
  128,192,160,192,3,5,5,4,0,0,192,32,96,32,192,4,
  5,5,4,0,0,160,208,208,208,160,3,5,5,4,0,0,
  96,160,96,160,160,3,4,4,4,0,0,96,160,160,96,3,
  5,5,4,0,0,96,128,224,160,64,3,4,4,4,0,0,
  224,192,160,192,3,4,4,4,0,0,224,128,128,128,3,5,
  5,4,0,255,96,160,160,224,160,3,4,4,4,0,0,64,
  160,192,96,4,4,4,4,0,0,176,96,96,176,3,4,4,
  4,0,0,224,64,32,192,3,4,4,4,0,0,160,224,224,
  160,3,5,5,4,0,0,160,64,160,224,160,3,4,4,4,
  0,0,160,160,192,160,3,4,4,4,0,0,96,160,160,160,
  3,4,4,4,0,0,160,224,160,160,3,4,4,4,0,0,
  160,224,160,160,3,4,4,4,0,0,64,160,160,64,3,4,
  4,4,0,0,224,160,160,160,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255};

olikraus, 2018-01-19:
This code got derived from a bdf file. For conversion you need the original bdf file.
It should have the extension .bdf.
You have this .bdf file?

phoenix7575, 2018-01-19:
Thanks for the answer, I do not have originals, I'll try to contact the author.
And you can not use it in this form?

olikraus, 2018-01-19:
> And you can not use it in this form?

No, it is like c code vs. generated binary code. Above is the binary code and we need the source...

olikraus, 2018-01-19:
> In the library there are very few Russian fonts there is only 3. 

There are much more than 3.

18 fonts are available here: https://github.com/olikraus/u8g2/wiki/fntgrpttyp0
Several Russian fonts are here: https://github.com/olikraus/u8g2/wiki/fntgrpnbp
Of course here: https://github.com/olikraus/u8g2/wiki/fntgrpunifont
and here: https://github.com/olikraus/u8g2/wiki/fntgrpcu12
and with a different encoding here: https://github.com/olikraus/u8g2/wiki/fntgrpcrox

olikraus, 2018-01-20:
I have create extra cyrillic fonts here: https://github.com/olikraus/u8g2/wiki/fntgrpx11
But these fonts will be available in the next stable release only.

closed 2018-01-20 by phoenix7575

phoenix7575, 2018-01-20:
Thanks for the work, we'll wait for the release. Only prompt please a line of compilation of a font bdf with a table  0-255. And how to properly use it, where ready to put. I want to make my own and how to properly connect I do not understand.

olikraus, 2018-01-20:
You do not need to wait for the next release. Many fonts are already available today.

If you want to create your own font: You must generate a .bdf file:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L179

You could, for example, use fony.exe (download here: http://hukka.ncn.fi/?fony) and export as .bdf.

olikraus, 2018-01-21:
You can download 2.21.5 beta release here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
install via "add zip lib" menu of the Arduino IDE.

phoenix7575, 2018-01-22:
Thank you, picked up what you need.
Only I did not Arduino IDE. Compiled in mingw32.