olikraus/u8g2 #487: ST7920 first line garbage

roberto314, 2018-01-19:
I've been porting u8g2 to ChibiOS ([http://www.chibios.org/dokuwiki/doku.php](url)) and i've managed to get a nokia 5110 display working (PCD8544 controller) but unfortunately the ST7920 via hardware SPI is not fully functional. The first line (about 16 pixel high) is distorted like issue #472. I tried different timings and different SPI speeds but no change. The problem is in "u8x8_d_st7920_common" . 
After trying a lot i figured the display always lets only ONE BIT CHANGE with a register write. So i just duplicated "u8x8_cad_SendCmd(u8x8, 0x03e );	/* enable extended mode */" and now it works.

olikraus, 2018-01-19:
where exactly did you apply the change? Maybe you can send me the updated code.

roberto314, 2018-01-19:
exactly where the line is. I just duplicated it.

olikraus, 2018-01-19:
you mean you duplicated this line:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7920.c#L128

roberto314, 2018-01-19:
That is correct. Now it works like a charm.
BTW. Is there a way to select fonts by number?
For now i made an array of pointer to font and can select like this:
u8g2SelectFont(&u8g2, fnt_ptr[1]);
Anything better?
So far i really like u8g2. I have some experience with uGFX but for monochrome u8g2 is much faster. Also font support is awesome.

olikraus, 2018-01-19:
Making your own array of pointers is a good idea and probably the only solution.

olikraus, 2018-01-21:
done, thanks!

closed 2018-01-21 by olikraus

roberto314, 2018-01-21:
Confirmed that it is the right place.
Thanks.