olikraus/u8g2 #488: Issue using "drawStr"

wEPac, 2018-01-21:
First, thanks a lot for this marvellous and fast driver, and all the job you did.

I was playing with it and fonts, I dont know if that s real issue but I noticed something weird. Drawing a text with "drawStr", the text was too long for the screen width, and the first character of the line came covered by something I suppose to be the characters from the truncated text.

Using "print" does not produce the same artefact, the truncated text is just ignored.

Best regards.

olikraus, 2018-01-21:
Will the problem vanish in 16Bit mode? I assume this is a 8 bit overflow.

wEPac, 2018-01-22:
hehe... yes! I only work in 8Bit mode at the moment, using g8u2 as it comes with an Arduino Nano.

olikraus, 2018-02-13:
ok, closing...

closed 2018-02-13 by olikraus