olikraus/u8g2 #489: ICSP and Hardware Constructor

wEPac, 2018-01-23:
Hello,

I tried a sketch with an Arduino Nano V3 and an OLED screen (128x64 monochrome) using the ICSP connector.

When I use the SW Constructor, everything works fine, but with HW Constructor, I see u8g2 sending 1st data, then waiting forever the answer from the Screen (I suppose, watching how the LED blinks 1 time then staying off).

I configured my OLED screen that way:

```
    U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, OLED_CS, OLED_DC);
```

with
```
   OLED_CS = 10;
   OLED_DC = MISO; // here MISO = 12 
```
and using the ICSP connector I got black screen, either connecting CS directly to pin D12.

trying with
```
   OLED_DC = 9;
```
everything connected on the ICSP, still a black screen, indeed, but now, connecting CS directly to the pin D9, everything is working.

So, it seems the DC/MISO cannot be read by u8g2 from the ICSP or did I make something wrong?

olikraus, 2018-01-23:
> So, it seems u8g2 cannot reads the DC/MISO from the ICSP or did I make something wrong?

I think this is a general problem of the atmega controller. It is NOT a u8g2 problem. See for example here: http://www.avrfreaks.net/forum/using-only-mosi-spi-and-using-miso-gpio-pin

wEPac, 2018-01-23:
Very interesting link, darn I looked for information all day long but could not find something neat like that, thank you.

I see we wont be able to use MISO tomorrow :) but nice, I clearly understand the problem now, and beside that, we can always use Hardware SPI and that is the most important.

olikraus, 2018-01-23:
:-)

closed 2018-01-23 by olikraus