olikraus/u8g2 #490: how to clear dedicate area buffer for SH1106 with u8g2_DrawStr?

wxiangcheng, 2018-01-24:
Hi, I'm using u8g2_DrawStr to display temperature periodiclly. How to update (x,y) area  before I do display? It may cause chaos after long time like '8' can't be updated with '1'. Thanks.
u8g2_ClearBuffer will update all OLED screen. What I want is only update (x,y) for several characters.
> u8g2_DrawStr(&iot_farm_u8g2, x,y, str);
	u8g2_SendBuffer(&iot_farm_u8g2);

olikraus, 2018-01-24:
Option 1:
Set the draw color (https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor) to 0 and erase the area below the string with drawBox() https://github.com/olikraus/u8g2/wiki/u8g2reference#drawbox

Option 2:
Set https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode to 0 
Along with font mode 0, you probably need to use monospace (or at least "h") font.
I think you will find all required information here:
https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode

Which font do you use at the moment?

olikraus, 2018-01-25:
For option 1, the size of the box depends on the string (of course).
For option 2 you probably want to print spaces after the strings, so maybe extend the string with space (assuming that str is a RAM buffer with sufficient size):
strcat(str, "  ");

wxiangcheng, 2018-01-26:
For option1, Can the area be cleared if I set correct size? Per my test, it doesn't clear but add some chaos image... Will try more.

For option2, sorry for my unclear clarification. The 'str' is what I want to display. My intention is  using  Spaces to do "clear" before DrawStr (My understanding). Would you please explain more about "you probably need to use monospace (or at least "h") font"?

olikraus, 2018-01-26:
Option 1:
Maybe you can send a picture for the "chaos"

Option 2: 

> My intention is using Spaces to do "clear" before DrawStr (My understanding).

This is not required. The new string will overwrite any existing content completly: This is the font mode 0. I think you should check the difference between font mode 0 and 1. 
With other words: In font mode 0, draw string does two things together: It clears the background AND draws the new string. However, if the new string is shorter than the existing string (on the background), then additional spaces ensure, that the background is completly erased.

"h" vs "t": The same font often exists as t and h variant. However this is not true for the chinese fonts. 
The difference is this: Let me assume, that there you wrote 
AAA
on the display
Now you want to overwrite this with in font mode 0
aaa
With the "t" font version the background box of the "a" glyph will not erase the captial A completly, so the upper part of "A" is still visible. This is not true with "h" version of the font. In the "h" fonts all gylphs have the same height (thats why the font is called "h"-font). This will ensure, that the previous text "AAA" is erased completly in font mode 0. As a result, you should use the h font in font mode 0 if you intend to overwrite existing background.

The drawback of the h font is, that it requires more memory than the t font.

Yet, because chinese chars are allmost all of the same height, the difference between t and h fonts is almost none, so I guess you can still use the t font for option 2. Of course I could also generate a h font if you require this.

AnHardt, 2018-01-26:
[Ucglibs HowToUseFonts.ino](https://github.com/olikraus/Ucglib_Arduino/blob/master/examples/HowToUseFonts/HowToUseFonts.ino)
is a nice example for what can go wrong, why, and how to fix - but for another library.

wxiangcheng, 2018-01-26:
what I test now is using number "1~10" and '.' and '-'. I found the tricky points:
1. when '1' is displayed firstly, then display '0', it is ok.
2. when '0' is displayed firstly, then display '1', not ok. Some part of '0' still remained on OLED screen.
3. Dot '.' and '-' can't  totally refresh other character because they are small. like firstly display "33", then display "1.8". I could see '3' still there.

I did not try Arduino SDK and  I can't 'fix my issue with above options. I need check more to demo before my customers :(. Appreciate if you can share more comments.

AnHardt, 2018-01-26:
Using a 'monospace' m ('not proportional') font will help here considerably.

wxiangcheng, 2018-01-26:
Thanks AnHardt. you found my root cause and sorry olikraus. I ignored your comments since I'm new to u8g2's font knowledge...  I could refresh it with MONOSPACE font. I have to generate new array with bdf tool "-b 2". I'm still looking for the proper font size but I think the original issue is gone. Thanks again.

olikraus, 2018-01-26:
It would be helpful to have a specific code. I could also create the missing font (it it is missing) with the monospace option (-b 2 or -b 1) and add this font permanently ot u8g2.

olikraus, 2018-01-26:
@AnHardt ... maybe I should create a similar example for u8g2.

wxiangcheng, 2018-01-27:
I fixed the issue now. Frankly, most users are not familiar with font related, like me. U8g2 is very good to use. But it cost memory especially for chinese font. I'm extracting my display characters from the BDF files.  For example, u8g2_font_unifont_t_chinese_My_customer, I only included 30 chinese characters plus ASCII 32-128. So the global array u8g2_font_unifont_t_chinese_my_customer size would be small enough. It would save my RAM. Is there any other convenient method to save more RAM QUICKLY? If there's please also share me. Thanks.

olikraus, 2018-01-27:
To save Ram, consider to use the picture loop approach of U8g2

wxiangcheng, 2018-01-27:
Do you mean like this for picture loop:
`void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,15,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}`

olikraus, 2018-01-27:
Yes, exactly

olikraus, 2018-02-13:
added full buffer font mode example to demonstrate the problems.

closed 2018-02-13 by olikraus