olikraus/u8g2 #491: 128x32 OLED screen SPI 4 wires with nrf51822 initialisation problem ???

Raidensama, 2018-01-25:
Hi, 
I'm working on a ssd1306 oled screen with the nrf51822 microcontroler and I think I've got initialisation problem judging from this picture
[image]

Raidensama, 2018-01-25:
If somebody knows what happens help me please. I think I'm really close to make this works.

fwachsmuth, 2018-01-26:
Try the SH1106. I had versions of this display that were sold as SSD1306 but actually had a SH1106 on them.

Raidensama, 2018-01-27:
Thanks for your reply, I tried it but it didn't work...

micooke, 2018-02-08:
Hey @Raidensama are you sure thats a 128x32? It looks like a 128x64

Raidensama, 2018-02-08:
No it's a 128x64 I made a mistake in the title...

micooke, 2018-02-08:
Hmm, post your initialisation routine please. Also, check that the pins are correct - particularly the reset pin. The nrf gpio pinmapping can be a bit wierd.. Oh yeah, and please confirm you are using the waveshare ble400 variant (assuming you are using sandeeps variants and ble core)

Raidensama, 2018-02-08:
This screen doesn't have reset pin... 
**This is how I start the program:**
u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8x8_HW_com_nrf51822,
            u8g2_nrf_gpio_and_delay_cb);		//Disply config
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2,0);

**And this is my function:**
uint8_t u8x8_HW_com_nrf51822(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{

	uint8_t *data;
switch(msg)
{

case U8X8_MSG_BYTE_SEND:
	 data = (uint8_t *)arg_ptr;

	 APP_ERROR_CHECK(nrf_drv_spi_transfer(&spi, data, 1, NULL, 0));
	

break;

case U8X8_MSG_BYTE_INIT:

		;
	nrf_drv_spi_config_t spi_config = NRF_DRV_SPI_DEFAULT_CONFIG;
				spi_config.ss_pin   = SPIM0_SS_PIN;
				spi_config.miso_pin = SPIM0_MISO_PIN;
				spi_config.mosi_pin = SPIM0_MOSI_PIN;
				spi_config.sck_pin  = SPIM0_SCK_PIN;
				spi_config.frequency = NRF_DRV_SPI_FREQ_1M;
				APP_ERROR_CHECK(nrf_drv_spi_init(&spi, &spi_config, NULL));

				nrf_gpio_cfg_output(DC_PIN);

				
				break;

case U8X8_MSG_BYTE_SET_DC:					//Pin Data/Comand

				nrf_gpio_pin_write(DC_PIN, arg_int);
				break;

default:
	return 0;
}
return 1;

}

Raidensama, 2018-02-08:
I tried it in debug mode and I saw that the bytes are sent in the right order but my screen displayed this... And the function to switch on the screen (u8g2_SetPowerSave(&u8g2,0)) works as you can see in the picture that means my SPI connections are good right?

Raidensama, 2018-02-08:
And Yes I'm using the ble400 waveshare

micooke, 2018-02-09:
```
data = (uint8_t *)arg_ptr;

 APP_ERROR_CHECK(nrf_drv_spi_transfer(&spi, data, 1, NULL, 0));
````
Just a note that data with its typecasting should not be needed

Also, i think it should be (note the tx length) 
``` APP_ERROR_CHECK(nrf_drv_spi_transfer(&spi, arg_ptr, arg_int, NULL, 0)); ```

I remember having something like this happen when i didnt set the reset pin correctly - but you don't have one so that's not the issue.

Although i dont think its the issue, what does your ```u8g2_nrf_gpio_and_delay_cb``` function look like?

micooke, 2018-02-12:
Did you try passing arg_int instead of 1 as per my previous message? If you did then Im all out of ideas, sorry :pensive:

Raidensama, 2018-02-18:
I tried this too... I'm waiting for another screen with a reset pin this time. And I will retry with this one.

micooke, 2018-02-18:
Goodluck :+1:

Raidensama, 2018-02-19:
Thank you @micooke

Raidensama, 2018-02-24:
I kept the changes you mentioned and I tried with an other OLED screen which has a reset pin and it works.
I add the reset pin in my function u8g2_nrf_gpio_and_delay_cb 
[image]

olikraus, 2018-02-24:
looks good :-)

micooke, 2018-02-24:
Awesome!

closed 2018-03-24 by olikraus

acharyanachiket, 2019-11-09:
Can you share your code and library. Because I am hard time with oled sh1106 who shows images at times and at times it's completely blurry ..Trying to figure out of its a reset pin issue of spi clk

micooke, 2019-11-09:
This is my screen test example
https://github.com/micooke/arduino-nRF5-smartwatches/blob/master/libraries/examples/Examples/u8g2_ScreenTest/u8g2_ScreenTest.ino

Sounds like you have a dodgy wire connection - likely the ground.

acharyanachiket, 2019-11-10:
Thank you let me have check the connections.