olikraus/u8g2 #492: "building" u8g2 for Arduino

fwachsmuth, 2018-01-26:
Hey Oli,

I forked your ingenious lib to make some changes in a more structured way than local patches – but am somehow stuck with turning the repo into a usable Arduino library. I found the create_release.sh in release/tools, but it seems to lack pieces.
Could you give me a hint of what the best way is to get library(.zip) loadable by the Arduino IDE out of this?

Thanks a lot!

olikraus, 2018-01-26:
I think you need to fork github project olikraus/U8g2_Arduino also. It must exist on the same level as u8g2 project. The create_release.sh script will then update the parallel U8g2_Arduino project.

Also note that a lot of code is generated.

 * If you add new devices, use https://github.com/olikraus/u8g2/blob/master/tools/codebuild/codebuild.c to update and regenerate the device related procedures
 * If you require new fonts, update https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c and rebuild the fonts: https://github.com/olikraus/u8g2/wiki/internal#rebuild-fonts

fwachsmuth, 2018-01-26:
That did the trick. Thank you so much!
I still had to do minor tweaks to the release script, but they are probably macOS-specific. I am happy now. :)

closed 2018-01-26 by fwachsmuth

olikraus, 2018-01-26:
:)