olikraus/u8g2 #495: [IL3820] Waveshare 2.9inch_e-Paper drawing issues

petrkr, 2018-01-27:
Hi,
I testing 2.9" e-Paper based on IL3820 and figured out some drawing issues. I tested library on OLED and it works fine, so I think there will be something regarding to e-paper one, I checked code, but I did not saw anything "at first good".

Problematic functions:
DrawFrame
DrawRBox
DrawDisc

Here are screenshost from GraphicsDemo sketch while using this constructor
`U8G2_IL3820_V2_296X128_F_4W_HW_SPI u8g2(U8G2_R0, /*cs=*/ D8, /* dc=*/ D2);`
_(I tested also without V2, and SW SPI). Also tested rotation R1 (much worst) and R2 (same, just 180 rotated)_

Tested platform:
MCU: ESP8266
SDK: 2.3.0
U8G2 lib: latest master (regarding to library version: 2.21.5)

Screens
[image]
[image]
[image]
[image]
[image]

Regards
Petr

petrkr, 2018-01-27:
Well...
my mistake, I missunderstood that note about `switching to 16-bit`. Also tried define that `U8G2_16BIT` in my sketch, but seems I need uncomment it in .h library, which I think is not so much good solution as I need to change library which I expect to be on all systems same (like if we are more developers). Can this be done some other way to control it from sketch code?

Petr

olikraus, 2018-01-29:
I always wished that Arduino IDE would allow extra options. Almost all other IDEs known to me have this option to pass extra defines to the compiler. Well it is not there with the Arduino IDE, so then the .h file has to be modified. I could permanently define this macro, but then AVR code will be 5% larger.

closed 2018-02-13 by olikraus

viveksoni95, 2019-03-19:
How to draw Bluetooth symbol on e-paper display ?

olikraus, 2019-03-19:
see  #841