olikraus/u8g2 #496: ra8835 128х128 not work?

githubrn, 2018-01-28:
pls help, not work hello world.

`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>

U8G2_RA8835_NHD_240X128_2_8080 u8g2(U8G2_R0, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11,12); 

void setup(void) {
  u8g2.begin();
  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}`

[image]

wired:

+5V LCD pins: 3,4,18
gnd LCD pins:  2,19

[vl-ps-mgcs128128-07.pdf](https://github.com/olikraus/u8g2/files/1671174/vl-ps-mgcs128128-07.pdf)

olikraus, 2018-01-28:
> +5V LCD pins: 3,4,18

According to the datasheet pins 4 and 18 need other voltage than 5V

githubrn, 2018-01-28:
I am just learning.
please tell me the table for pin 4 and 18.

Cold cathode fluorescent lamps (CCFL) backlight. +12V through the inverter - it works. PCB inverter separately.

Thank you!

olikraus, 2018-01-28:
> Cold cathode fluorescent lamps (CCFL) backlight. +12V through the inverter - it works.

Then the datasheet is wrong: Table 6 requests 235V.

According to datasheet pin 4 requires -17 V

githubrn, 2018-01-28:
maybe option

[image]
[image]

it?
[image]
why -17V?

olikraus, 2018-01-28:
yes, should be +17V.

githubrn, 2018-01-29:
Thank you.
ordered a voltage regulator.
I'll write a result later.

olikraus, 2018-03-24:
feel free to reopen this issue for further discussion...

closed 2018-03-24 by olikraus