olikraus/u8g2 #497: Good Clue for more Space - Kill no need Font Symbols

bobybobybob, 2018-01-28:
Dear Codemaster and Friends from u8g2

RAM ROM FLASH Space is small and some Symbols need much RAM ROM FLASH. 

De schets gebruikt 31958 bytes (**99%**)  programma-opslagruimte. Maximum is 32256 bytes.
Globale variabelen gebruiken 1500 bytes (73%) van het dynamisch geheugen. Resteren 548 bytes voor lokale variabelen. Maximum is 2048 bytes.

I have good Clue to make the used Fonts again with hints and tolls from Codemaster. I have aditional use "FONTFORGE" for Windows and open TTF and remove all no need symbols. 

This website help me to find only the need symbols http://www.characterfrequencyanalyzer.com/english/index.php

otf2bdf -r 72 -p 16 mysweetfont.ttf -o mysweetfont.bdf

ptf2bdf need linux. I install it with command apt-get install otf2bdf     and all was wonderfull. 
(-p 16 is real fontsize about 11   i use formula  comandlinefontsize = 100:78*arduinofontsize  )

and after get my mysweetfont.bdf

bdfconv -v -f 1 mysweetfont.bdf -o mysweetfont11.c -n mysweetfont11.c

After i integrate the mysweetfont c Files  in my project with just paste it in ino window i compile again and perfect !!! New Space is free for more Code and u8g2 work perfect too.

De schets gebruikt 22804 bytes (**70%**)  programma-opslagruimte. Maximum is 32256 bytes.
Globale variabelen gebruiken 1500 bytes (73%) van het dynamisch geheugen. Resteren 548 bytes voor lokale variabelen. Maximum is 2048 bytes.

olikraus, 2018-01-28:
So you created your TTF font by removing the unused chars?
There are also alternative methods: 
 * You could use a text editor and remove the gylphs from the bdf file itself (bdf is a text document).
 *  Use the -m option for bdfconv. Here are some examples:
```
  -m '32-255'     select gylphs from encoding 32 to 255
  -m '32-255,~64' select gylphs from encoding 32 to 255, exclude '@'
  -m '32,48-57'   select space, '1', '2', ... '9'
```

bobybobybob, 2018-01-29:
Hello,
Yes i open TTF with Fontforge Freeware and clear all no need and then save as new TFT. Also is possible to mix fonts with funny Fonts like Wingding.

With little understanding u8g2 is realy the best display driver and it make fun to get idea from brain to Computer. Thank you very much Codemaster that you make that possible.

olikraus, 2018-01-29:
:)

closed 2018-02-12 by bobybobybob