olikraus/u8g2 #501: 128x128 OLED I2C using SH1107 driver

dominicklee, 2018-02-04:
Hello all, I have an OLED that uses the SH1107 controller.  I saw that the u8g2 library supports the SH1107 controller but it only offers a resolution of 128x64 in a couple cases.  I would like to use this library for my 128x128 OLED display that uses SH1107 via the I2C pins of an ESP board.  At the same time, I would like to run another I2C module such as the MPU6050 simultaneously.

If anyone could help or provide a detailed suggestion, I would appreciate it. To sum it up:

- Need support for 128x128 OLED with SH1107 via I2C
- Need support for this OLED while using another I2C device (such as MPU6050)

Thank you

olikraus, 2018-02-04:
This is related to #418 

I think support is only there in the beta release. 
Can you download U8g2 2.21.6 from here and test with this version:
https://github.com/olikraus/U8g2_Arduino

Install via Arduino IDE (add zip lib menu). How will the 128x128 display look like with the 128x64 driver?
Can you attach a picture?

dominicklee, 2018-02-04:
Okay, I have followed your advice and downloaded U8g2 2.21.6 from your repository. I have installed the library and selected the following constructor:

`U8G2_SH1107_64X128_2_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);`

Then I opened the GraphicsTest example and uploaded the code.

Originally, I have tried to select the `SW_I2C` constructor but for some reason I end up with function reference errors during compilation. The `HW_I2C` constructor works and the display lights up but I am seeing only 64 pixels x 128 pixels. As shown in my picture, everything on the OLED display looks very narrow.

So basically, I think there needs to be a similar profile for SH1107 that would support the 128 x 128 pixel display. Can you please help me add the code needed? Please let me know and I would be glad to test your codes with my 128 x 128 OLED.

Thanks!

[image]
[image]
[image]
[image]

olikraus, 2018-02-04:
> Originally, I have tried to select the SW_I2C constructor but for some reason I end up with function reference errors during compilation. 

Strange. Which Arduino Board?

> So basically, I think there needs to be a similar profile for SH1107 that would support the 128 x 128 pixel display. 

Of course. But i wanted to see to which extend the 64x128 device works. It is usually better to start with something, that works already a little bit ;-)

dominicklee, 2018-02-04:
I'm using NodeMCU v1.0 (Esp8266) as the board with D2 and D1 as the hardware I2C.

I hope my insights help will bring us close to getting this working on the 128x128 OLED. I'm looking forward to it  :D

olikraus, 2018-02-04:
any datasheet for your OLED?

olikraus, 2018-02-04:
I have create a new release 2.21.7 here: https://github.com/olikraus/U8g2_Arduino

Can you test the FlipMode Example from the page mode examples?
Use constructor 

```
U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
```

Is the content still shifted? Can you sent a picture of the first flipmode picture?
How das the second flip mode picture look like?

dominicklee, 2018-02-04:
Thanks. I have downloaded 2.21.7 and uploaded the FlipMode Eample from page mode and used the constructor as you mentioned. 

It seems that the content is still shifted. Please see my attached photos of the first, second, third, and fourth flip mode screens. I am not sure why the display is shifted.

[image]
[image]
[image]
[image]

By the way, I know you wanted the datasheet for my OLED, so I have attached it:
[SH1107V2.1.pdf](https://github.com/olikraus/u8g2/files/1693181/SH1107V2.1.pdf)

I hope the datasheet might be insightful.

olikraus, 2018-02-04:
Thanks for the controller datasheet. Yet, i was more interested in the display datasheet. It looks like that there is a 32pixel offset, which I do not really understand. My hope is, that there is some information about the 32pixel offset in the display datasheet.

olikraus, 2018-02-04:
ok, let's try to fix the offset problem. Please locate file u8x8_d_sh1107.c on your disk.
Please put line 341 in comments: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L341

What will happen. Will this help?
If this doesn't help, please update
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L374
and
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L375
to:
```
  /* default_x_offset = */ 96,
/* flipmode_x_offset = */ 96,
```

Will this help?

dominicklee, 2018-02-04:
I have tried to comment out line 341 in: [https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L341](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L341)

However, that did not change anything. But I followed your advice to change lines 374 and 375 in [https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L374](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L374)
to:
```
/* default_x_offset = */ 96,
/* flipmode_x_offset = */ 96,
```
And then the frame aligned perfectly!  Thank you very much for your help. Please see my pictures attached for reference. If there are any other tests you would like me to run, please let me know.

[image]
[image]
[image]
[image]

Also, is there a way you can fix the `SW_I2C` constructor for this OLED so that I could use this I2C display alongside another I2C module such as an IMU?  Thank you again!

olikraus, 2018-02-04:
Great, thanks for testing.

The SW_I2C constructor should work. Can you connect your display to some other pins. Of course you need to name the pins to the constructor.

olikraus, 2018-02-13:
looks good so far. I think this can be closed. Let me know of any problems.

closed 2018-02-13 by olikraus

dominicklee, 2018-02-18:
Hi @olikraus, I notice a slight flicker in my OLED. Also, when I try to record the OLED display with my smartphone, I see fast scrolling lines.

Is there any values I can tweak to increase the refresh rate? Thanks.

olikraus, 2018-02-18:
hmm, you could locate file u8g2_d_sh1107 (https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1107.c#L342) and modify the following command:
Instead of U8X8_CA(0x0d5, 0x051)
try:
U8X8_CA(0x0d5, 0x050)
will this be better?
What about 
U8X8_CA(0x0d5, 0x070)
?

dominicklee, 2018-02-18:
Thank you!  I tried using both U8X8_CA(0x0d5, 0x050) and U8X8_CA(0x0d5, 0x070). It seems that U8X8_CA(0x0d5, 0x050) gave a better result.

Just for context, I am running this U8g2 on the ESP8266. On my OLED, for some reason I always see "vertical bars" on my image being displayed. The bars stay in the same spot, and are not flickering. Could you please tell me what can be tweaked to reduce or remove those bars?

Please see my attached image to see what is seen on the display.

[image]

olikraus, 2018-02-18:
Ideas:
 * Place a 1uF cap between power supply and GND, close to the display
 * Reduce contrast
 * Change values for the following two commands:
```
  U8X8_CA(0x0d9, 0x022), 		/* [2] pre-charge period 0x022/f1*/
  U8X8_CA(0x0db, 0x035), 		/* vcomh deselect level */  
```

microbirdman, 2018-03-05:
Hi @dominicklee 

Amazing work! I've got a SH1107 compatible display but I can't make it work.
I am getting a black screen no matter what I do :) This is the first time I am trying to use an OLED
display that doesn't come with a breakout circuit. I am pretty sure I am missing something.
I think everything is wired properly but the VPP. Where do you connect the VPP?
Sorry if the question is too lame but I haven't seen this particular issue being asked yet.
Thanks in advance.

dominicklee, 2018-03-08:
Hi @microbirdman can you please share a picture of the schematic you've used?

Also, please share a picture of your connections. I'm not sure with how you hooked it up. There are some power boost and line regulators that are required.

olikraus, 2018-03-08:
@dominicklee did you improve the quality of your screen? Which (if any) of my suggestions did help?

dominicklee, 2018-03-08:
Yes, I added the 1uF cap. Actually, I increased the contrast and noticed improvements.

dominicklee, 2018-03-08:
Currently, the imperfections on the screen have become barely noticeable. If you look really hard, it is still visible, but much better.

dominicklee, 2018-03-08:
@olikraus I actually tried everything you suggested. To conclude, I see less lines when I increased the contrast instead of reducing.

Also the more I deviate from the POR level values, either no change or the more worse it becomes. I say changed the value of Contrast Data Register from default 0x80 to 0xFF and that increased contrast and hence reduced the lines.

dominicklee, 2018-03-08:
Yes. The lines were only reduced to be less visible. But the quality remains unchanged.

olikraus, 2018-03-08:
hmmm. so even a ceramic cap did not improve anything?

microbirdman, 2018-04-09:
Hi @dominicklee ,

Thanks very much for your reply and sorry for the delayed answer. I didn't get the notification on my email but it's sorted now. I am following the schematic attached for the screen. I connected 3.3v to VDD and an external VPP (power supply I had at home) of 13.3v aprox. sharing same ground but it didn't work (black screen). Do you use an IC to do that voltage boost? Any suggestions? Thanks in advance.

[image]

dominicklee, 2018-06-02:
@olikraus Sorry for the late reply! Again, I am bringing up the flickering issue. The display still has noticeable flickering lines. These lines are still visible to human eye but especially visible when filmed with a camera. 

I am running your code on an ESP8266 if that context makes any difference. I feel it has something to do with the frequency of the display refresh or related to the speed. We need to make the speed of the display faster to prevent this effect being noticed by human eye.

Please see the videos I recorded of the display. I filmed a Normal speed video and slow motion version to show you what it is doing. Please give me insight to what problem this might be. Any suggestions on what can be tweaked in the code would be appreciated!

[OLED flickering videos.zip](https://github.com/olikraus/u8g2/files/2064690/OLED.flickering.videos.zip)

Thanks again!

gavan1, 2018-08-15:
Could you post a simple outline / link of updated library for the 128x128 and best settings?

Thanks

olikraus, 2018-08-15:
Latest version is here:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).