olikraus/u8g2 #502: Font Request

LionheartUK, 2018-02-04:
Thank you for creating such a useful library!

Is it possible to add Profont pixel heights of 58 and 61?

Regards

Richard

olikraus, 2018-02-04:
sure, if you can provide the related font files (bdf file format prefered)

olikraus, 2018-02-05:
I do not remember details, but in general profont is a bitmap font. The current fonts for U8g2 are just based on the bdf bitmap fonts for some specific sizes: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf

LionheartUK, 2018-02-05:
I have found the TTF font file, I know you said your prefer BDF (working on that..)
[ProFont.zip](https://github.com/olikraus/u8g2/files/1694731/ProFont.zip)

olikraus, 2018-02-05:
ok, looks promising. I also need the source of the zip file (due to required copyright statements...)

LionheartUK, 2018-02-05:
no problem! it came from http://tobiasjung.name/profont/

olikraus, 2018-02-05:
ok, great, so I wait for further input...

LionheartUK, 2018-02-05:
Ive tried, but fonts just are not my strong point and I have no idea how to provide you with a BDF version of the font! Thanks for your help anyway.

olikraus, 2018-02-05:
Basically BDF conversion is simple, either use otf2bdf or fontforge. The problem is more, that I would "just do the conversion". I can not modify the result or do any kind of manual postprocessing. I also do not know what you mean by height of 58 and 61.
I also do not know the quality/state of the profont.ttf. Maybe the generated glyphs look very ugly.

The problem is, TTF usually is a vector format which renders graylevel bitmaps on the fly. U8g2 requires precalculated black/white bitmaps (such as BDF format). Blindly converting vector fonts may end up in very bad results. Thats why I hesitate a little bit to include converted fonts... although i did in history with logisoso and other fonts... but it took me days to get suitable results. 

The point is this: I am happy to add new bitmap fonts, but I do not want to work on the rendering problem of TTF fonts.

olikraus, 2018-02-13:
no further response, closing...

closed 2018-02-13 by olikraus