olikraus/u8g2 #503: KS0108_ERM19264 not selecting chips correctly

l0ner, 2018-02-08:
As it says in the title

on the ERM19264 LCD CS pins and reset pins are active LOW, instead of being active high. ([Datasheet](http://www.displayfuture.com/Display/datasheet/monographic/ERM19264-1.pdf))

Right now:
When the library tries to write to the first chip, is sets the CS0 high and the other two low, thus actually selecting chips 2 and 3.

How it should work:
Set the CS0 low, CS1 to high and CS2 to high in order select the first chip. Same for the other chips.

olikraus, 2018-02-08:
hmm.. I try to track down the problem. I think I stated this here correctly:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ks0108.c#L214

Is is then received by the byte processor here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_byte.c#L358

Then it is processed here:
```
void u8x8_byte_set_ks0108_cs(u8x8_t *u8x8, uint8_t arg)
{
  u8x8_gpio_SetCS(u8x8, arg&1);
  arg = arg >> 1;
  u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_CS1, arg&1);
  arg = arg >> 2;
  u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_CS2, arg&1);
}
```
Finally the gpio level is set here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L146

Still I do not see that the signals are inverted...

So from my perspective u8g2 behaves as it should. So the question is... Do you use the original u8g2 code? Which is your board? Did you download u8g2 via the Arduino IDE?

olikraus, 2018-02-13:
no response, closing

closed 2018-02-13 by olikraus