olikraus/u8g2 #504: add initialize option for redirecting display output

cyberman54, 2018-02-10:
If programming for different type of boards with and without OLED display it would be great to have an option to call u8g2 / u8x8 with an constructor which does nothing. This is useful on boards without OLED display, driving them with the same code that also runs on board with OLED displays.

cyberman54, 2018-02-11:
thanks!
should be implemented both for u8x8 and u8g2.

olikraus, 2018-02-17:
u8x8_setup.c 
```
uint8_t u8x8_d_null_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
```

u8g2_setup.c
```
void u8g2_Setup_null(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
```

olikraus, 2018-02-17:
new devices:
```
U8X8_NULL u8x8;
U8G2_NULL u8g2(U8G2_R0);
```

some testing in the Arduino IDE required...

cyberman54, 2018-02-18:
Could you release a new version with this feature to platformio lib?

olikraus, 2018-02-18:
> Could you release a new version with this feature to platformio lib?

unfortunately no. I have no idea about platformio. But i will release u8g2 soon into the Arduino IDE.

olikraus, 2018-02-18:
test ok...

closed 2018-02-18 by olikraus