olikraus/u8g2 #505: cannot address 320x240 screen in U8G2_RA8835_320X240_1_8080 mode

phe78, 2018-02-10:
Hello, I am using a RA8835 320xx240 screen with a 8080 interface (model reference https://www.buydisplay.com/default/5-7-inch-320x240-lcd-display-graphic-module-touch-panel-black-on-white ). It appears the adressable screen stays limited to a 240x240 window. Any attempt to adress X>240 is wrapped to zero. 
I am using an Arduino Nano and cannot use the Full buffer mode as the RAM amount is limited. 
Am I doing something wrong, or maybe there is a problem is the screen buffer management ?
thanks in advance for your feedbak
Philippe

JamesGKent, 2018-02-10:
In the faq you'll find a bit about enabling 16 bit mode, you need to do that to address screens larger than 256 pixels

phe78, 2018-02-10:
thanks for your quick response James. My display only supports 8-bit 8080 interfacing.

olikraus, 2018-02-10:
You must enable U8g2 16 Bit. It has nothing to do with the interface. It just enables 16 bit arithmetics fo x and y coordinates.

phe78, 2018-02-11:
thanks a lot. worked perfectly.

closed 2018-02-13 by olikraus